sebanc/brunch #702: Lenovo thinkpad t430 - mostly working - some issues with touchpad

Lucaacer, 2020-12-11:
Thanks as usual for still supporting this project.

I installed chrome OS through your framework on the ssd of my Lenovo Thinkpad t430, 8 gigs ram and I5 3320M cpu and everything work fine.

Triple boot windows 10, MX Linux and chrome OS.

It took me some time to understand that I was unable to start chrome OS from MX Linux's grub since there was no /boot/efi partition. For some reason MX Linux was defaulted on a non efi installation or maybe it had been a mistake by me.

Anyway, I have got a very small usb key, so slim that it can stay always attached, with a working install of chrome OS and I edited its grub and added a menuentry pointing at /dev/sda4. It's a small work-around and it took just a few minutes but, as soon as I have some more spare time, I will install ubuntu in uefi mode to solve this small issue.

Performances are really very, very good and much better than running chrome OS off an usb key. Linux apps are very fast and that's the main difference comparing to a N4100 cpu. I mean, browsing is very fast even with the latest celeron cpus, but linux app in crostini are somehow slow with such cpus (much slower than running GNU Linux on bare metal) while the i5 of this thinkpad delivers very good performances.

The only issue is with touchpad:

- there is no way of zooming in or out (I have got to use ctrl + and ctrl - );

- swiping with 2 fingers works, even for scrolling, while no 3 fingers gesture works;

- the worst thing is that the feeling of the touchpad is very weird and, when you release your finger, the cursor sometimes jumps a bit far from its destination, as if it were a living thing... Yes, it's a little bit funny.

Let's say that, with some adaptation, the touchpad can be used, but it somehow spoils the general experience which is great.

Have you got any idea on a way of fine tuning the behavior of the touchpad?

Thanks in advance.

sebanc, 2020-12-12:
Have you tried both "alt_touchpad_config" and "alt_touchpad_config2" options ?

Also, for touchpad gestures with more than 2 fingers, have you tried adding "psmouse.synaptics_intertouch=1" ?

Unfortunately, touchpad issues are very difficult to debug when you don't have the device, some of the custom configurations mentioned in #305 would probably help but you will have to try to identify the correct one for your device.

Lucaacer, 2020-12-15:
Thanks, I unsuccessfully applied such options, but nothing changes.

To be true, 3 fingers gestures do not work in Windows 10 as well, so I did not expect them to work in Chrome OS. 

2 fingers zoom instead works in Windows, so it might somehow work in Chrome OS too.

Regarding the excessive sensitivity of the touchpad, I reduced its speed and disabled acceleration and now it's more or less usable, even because I have got accustomed to its behaviour, 

You may close this issue.

Lucaacer, 2020-12-15:
These changes made my touchpad less sensitive and thus better

[40-touchpad-cmt.conf.txt](https://github.com/sebanc/brunch/files/5698751/40-touchpad-cmt.conf.txt)

sebanc, 2020-12-16:
Sorry but I see no change compared to the standard config, could you highlight your changes or post a diff ?

Lucaacer, 2020-12-16:
I edited these values, by making them higher, and the cursor has now lost the tendency to "escape" when I release my finger.

Regarding drumroll supression, I actually cannot say if it makes any difference, to be true.

Option          "Drumroll Suppression Enable" "1"

  # Enable Stationary Wiggle Filter
    Option          "Stationary Wiggle Filter Enabled" "1"
    Option          "Finger Moving Energy" "0.0028"
    Option          "Finger Moving Hysteresis" "0.0044"

    # Avoid accidental scroll/move on finger lift
    Option          "Max Stationary Move Speed" "100"
    Option          "Max Stationary Move Speed Hysteresis" "2"
    Option          "Max Stationary Move Suppress Distance" "0.8"

sebanc, 2020-12-26:
Sorry for the delay, I was going to add this config when I noticed something weird, the changes you made should only apply to touchpads with USB ID 05ac:0265 or 004c:0265.

Could you attach the output of `lsusb` in crosh shell ?

Lucaacer, 2020-12-28:
Sorry, wrong edits:)

iafilius, 2021-04-14:
Hi, 
Hi,
Hi, just installed brunch 20210403 and samus 89 recovery image on a lenovo T430/8GB Ram and having:
- Touchpad issues (working properly with other OS)
- Lower key part of touchpad, not working (working properly with other OS)

Tried with edit_grub-config from root shell via crosh the advised: "alt_touchpad_config" / "alt_touchpad_config2". Noticed a difference (in speed), but same issues shows up.

The touchpad seems 
- to ignore short fast movement
- slow long movement are detected
- The habit of jumping a bit back
- The mouse key on keyboard works as expected, same as the mouse keys part of keyboard.

Tried the “crouton” app (demo app from appstore), but don’t see any special, perhaps tried wrong program?

Some information from the system while running brunch/chromos
```
 # grep -i touchpad /var/log/messages
2021-04-14T18:22:17.922548Z INFO kernel: [    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=alt_touchpad_config console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
2021-04-14T18:22:17.922891Z NOTICE kernel: [    0.195862] Kernel command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=alt_touchpad_config console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
2021-04-14T18:22:17.927282Z WARNING kernel: [   36.554929] brunch: /rootc/patches/61-touchpad.sh success
2021-04-14T18:22:20.960036Z INFO kernel: [   43.211280] psmouse serio1: synaptics: Touchpad model: 1, fw: 7.2, id: 0x1c0b1, caps: 0xd047b1/0xb40000/0xa0000/0x0, board id: 0, fw id: 920262
2021-04-14T18:22:20.999032Z INFO kernel: [   43.250533] input: SynPS/2 Synaptics TouchPad as /devices/platform/i8042/serio1/input/input15
2021-04-14T18:29:28.937037Z INFO kernel: [    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=alt_touchpad_config2 console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
2021-04-14T18:29:28.937382Z NOTICE kernel: [    0.195337] Kernel command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=alt_touchpad_config2 console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
2021-04-14T18:29:28.944548Z WARNING kernel: [   36.573248] brunch: /rootc/patches/61-touchpad.sh success
2021-04-14T18:29:32.086357Z INFO kernel: [   43.336333] psmouse serio1: synaptics: Touchpad model: 1, fw: 7.2, id: 0x1c0b1, caps: 0xd047b1/0xb40000/0xa0000/0x0, board id: 0, fw id: 920262
2021-04-14T18:29:32.126028Z INFO kernel: [   43.376322] input: SynPS/2 Synaptics TouchPad as /devices/platform/i8042/serio1/input/input15

```

Lsusb:
```
# lsusb 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 04f2:b2da Chicony Electronics Co., Ltd thinkpad t430s camera
Bus 001 Device 003: ID 147e:2020 Upek TouchChip Fingerprint Coprocessor (WBF advanced mode)
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
localhost ~ # 

```

If desired/usefull I can deliver same/additional info created from working Linux (Fedorra or ubuntu)

Arjan

polocatfan, 2025-01-28:
> I edited these values, by making them higher, and the cursor has now lost the tendency to "escape" when I release my finger.
>
> Regarding drumroll supression, I actually cannot say if it makes any [...]

this did nothing.

polocatfan, 2025-01-28:
> Sorry for the delay, I was going to add this config when I noticed something weird, the changes you made should only apply to touchpads with USB ID 05ac:0265 or 004c:0265.
> 
> Could you attach the output of `lsusb` in crosh shell ?

lsusb gave me an unknown command error

closed 2026-03-31 by github-actions[bot]