sebanc/brunch #703: Tablet mode not working on Brunch r86 k5.4 stable 20201201

ashnazir786, 2020-12-11:
I updated my Elite book G5 1080 and tablet mode has stopped working. I reinstalled with Chrome OS with latest build still does not work. Reinstalled with previous version and now it is back. Thanks for all your help.

sebanc, 2020-12-12:
It seems there is a regression wrt tablet mode on some devices with 5.4 kernel. For now, I have not identified what's causing this but the future brunch releases will have both 4.19 and 5.4 kernel included.

If you want to try it, a test build with both kernels has been uploaded in "brunch-unstable" repository. (5.4 is the default, if you want to boot with 4.19 kernel replace "/kernel" on the grub configuration linux line with "/kernel-4.19")

librehat, 2020-12-16:
I ran into a slightly different issue. Upgraded to r87 with kernel 5.4, my device is now booted into tablet mode by default, and changing the variable in `/sys` doesn't make any difference (to exit the tablet mode). 
Device: Lenovo YOGA 700

sebanc, 2020-12-16:
@librehat Could you try adding "module_blacklist=thinkpad_acpi" on the kernel command line ?

sebanc, 2020-12-16:
@ashnazir786 Could you try adding "module_blacklist=thp-wmi" on the kernel command line ?

librehat, 2020-12-16:
@sebanc It was very weird so I first rebooted with kernel-4.19 and it worked fine then I rebooted again and the GRUB couldn't boot anymore. So I changed it to kernel-5.4 and so far it's been working fine?? I'm not sure what is going on but the tablet mode issue is also gone

maxdojin, 2020-12-16:
Tablet mode worked until r86 k4.19 stable.  Tablet mode does not work with kernel 5.4 4.19 on r87 stable
 The model is Panasonic Let'snote mx-5
Thank you

mrwiwi, 2020-12-21:
Same here on latest chromeOS rammus and latest brunch, everything works so far on my Dell 5290 2in1

sebanc, 2021-01-01:
If you still have this tablet mode issue could you run the below commands in crosh shell and post the output:
- `lsmod | grep intel_vbtn`
- `cat /sys/class/dmi/id/chassis_type`

Also please attach a dmesg (run `dmesg > ~/Downloads/dmesg.txt` in crosh shell and attach dmesg.txt from your downloads folder)

maxdojin, 2021-01-02:
I'm using brunch_r87_unstable_20210101
intel_vbtn             16384  0
chassis_type 10
[dmesg.txt](https://github.com/sebanc/brunch/files/5759805/dmesg.txt)

sebanc, 2021-01-03:
@maxdojin Could you try the latest build in the brunch-unstable repo ?

maxdojin, 2021-01-03:
I tried using Brunch r87 unstable 20210102
Tablet mode is working perfectly
Thank you

sebanc, 2021-01-03:
Thanks for the feedback :)