sebanc/brunch #704: without uefi may run chromeos?

xiaomayi755259, 2020-12-12:
[image]
please help!  install chromeos and run it just like this. my computer without uefi. install chromeos depend on mbr mode. install it in usb. if solved it may i install it in hdd? thanks help

>

glowingnec, 2020-12-12:
You should wait for a little bit, the first boot could be quite longer ( mine take ~ 10 minutes )
Also, did you install with `mbr_support.tar.gz` patch?

sebanc, 2020-12-12:
1) unpack the brunch archive
2) unpack the mbr_support.tar.gz archive in the same folder (say yes to overwrite the files)
3) follow the rest of the guide

Note: Make sure your laptop is at least intel Gen 1 as indicated in the readme

xiaomayi755259, 2020-12-12:
the step one. two and three just like the files describe.  but my computer is intel(R) core(TM) duo cpu T8300.  if install chromeos in usb don't run. may i install it in win10?  thanks

sebanc, 2020-12-12:
Sorry but core2duo cpu are not supported as explained in the main Readme