sebanc/brunch #705: Successfully Dual Boot ChromeOs and Windows 10 in legacy BIOS / MBR without installing Linux OS

godhek, 2020-12-12:
Hi all!
First of all, I'd like to thank @sebanc and other contributors here for their great work!
I just want to share my success story in installing brunch in legacy BIOS / MBR system after countless hours of browsing, reading and experimenting. I don't know whether this kind of guide had been posted here before, but I just thought of sharing with all the users here, and hopefully can benefit others.  

Next, I'd like to clarify that I'm not a programmer nor software engineer. Thus, I have very limited knowledge in programming and Linux commands. So, please don't ask me about booting command lines since I, frankly, don't know and might give the wrong answer. I just copy and follow guides and commands suggested by others.  So, it just luck plus trial and error :) I only consider myself as a computer / tech enthusiasts. Thus, I don't know and not responsible if any trouble might arise resulted from attempting this installation.

My pc is a Dell Inspiron 2320 AIO with 2nd gen core i7 CPU, 2TB of HDD with Windows 10 installed, touch screen and 8G of RAM from around 2011. It doesn't have UEFI option in its BIOS, and I leave the HDD as it is because I don't want to damage its original recovery partition inside.

Briefly, these are general steps taken that I remember:
- Create Rammus image inside an 16GB USB flash drive with the MBR zip as well by following @sebanc 's guide using Ubuntu WSL.
- Try booting the flash drive to check whether it works and don't forget to resize data and reboot again.
- Unplug it and boot to Window to create 16.2GB NTFS logical partition in an existing extended partition using Windows Disk Management, without assigning any drive letter, named it Chrome. If you created just 16GB of space, it would later on gave an error of not enough space when installing the ChromeOS image with size of 16GB to the hard drive.
- I got the idea about this Grub2 from [AIOBOOT](https://www.aioboot.com/en/install-grub2-from-windows/) and download it from [here](http://ftp.gnu.org/gnu/grub/grub-2.02-for-windows.zip). I failed to install Grub2Win, which frequently recommended. Maybe because of not having UEFI in my pc.
- Follow the guide to Install i386-pc for BIOS / MBR option 
- Open and edit grub.cfg using notepad with administrator privilege and add Windows booting command that I found from somewhere posted on Stack Exchange at the last part of the file as follows:

`menuentry "Windows 10" --class windows --class os {
insmod ntfs
set root='(hd0,2)'
ntldr /bootmgr
}`
- Restart and try to boot to Windows (I tried and error for that number '2' which means 2nd partition).
- If successful, edit and save that grub cfg file again to correct the Windows partition number and can continue to follow @sebanc 's guide in dual booting ChromeOs from HDD to the NTFS partition created previously and it's named Chrome. Until step number 6.
- Note (or even take a photo) of that boot configuration file after successfully installed the ChromeOs.
- Reboot to Windows to edit the grub.cfg using notepad with administrator privilege and carefully rewrite the ChromeOs boot configuration after Window entry at the bottom, without the rmmod tpm line.
- Add the following lines to that configuration entry, which I got the idea in one of @sebanc 's mention about Arch Linux and googled it

`menuentry "ChromeOs" {
	insmod ntfs
        insmod cpio
        insmod part_gpt
        insmod part_msdos
	set root='(hd0,5)'
	img_path=/chromeos.img
        img_part=/dev/sda5
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_path=$img_path img_part=$img_part
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}`
- After searching and reading from somewhere on [Stack Exchange](https://unix.stackexchange.com/questions/508617/grub-boot-iso-disk-image-from-ntfs-partition), looks like Arch Linux has 'mkinitcpio' which enable its grub to read / 'hook' to contents and partitions from img file (CMIIMW...), and fortunately this Grub2 has cpio module as well. I also added insmod part_gpt, and insmod ntfs because the file system type where it is installed and partition type used inside the image. I added insmod part_msdos because the overall partition type of this system as when you type ls command in grub prompt.
- Save the cfg file and try booting to ChromeOs. It will take a while when booting for the first time, so as long that hard drive still blinking, just wait patiently until greeted by ChromeOs logo and initial set up page.
- You can also add kernel option such as console=, enable hyperthreading and  ACPI power button option once everything is working to the cfg file.

An honorable mention as well to [ETA PRIME](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiH8fur48jtAhVQbn0KHb0eBpkQtwIwAHoECAUQAg&url=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DROBpXNonVxc&usg=AOvVaw01UBhuyuob2F3SxHcnTbQD) and [Kedar Nimbalkar](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiH8fur48jtAhVQbn0KHb0eBpkQtwIwBHoECAQQAg&url=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3D4gZYV0RWJQ8&usg=AOvVaw39DYeKgDOdyE-QkS51O17H) on YouTube whom gave me the idea of installing ChromeOS.

Cheers!

sebanc, 2020-12-13:
Thanks for posting the detailed guide :)

godhek, 2020-12-14:
> Thanks for posting the detailed guide :)

Don't mention it. It's nothing compared you and other contributors have accomplished in this project so far. Cheers!

Anyway, most Google services are down at this moment!

natri23, 2020-12-30:
@sebanc so I magaged to Dual boot Brunch with a BIOS only laptop as well.
At first it didn't work, threw error: no partition found.  And I realized that grub2 doesn't read GPT partition table by default
So I added `insmod part_gpt` to the grub menuentry and it works like a charm.
You might want to add it to chromeos-install.sh

simplyrajatgupta, 2021-05-31:
Try this!!!
It will absolutely work
**[https://github.com/sebanc/brunch/issues/1088]()**