sebanc/brunch #706: Stuck at boot on HP Elitebook 2560p

OlivierPonel, 2020-12-13:
Hi everyone,

First of all, thank you @sebanc and all other people involved in this great piece of software !!

I need some help because i didn't managed to install ChromeOS on my old laptop.

It's an HP Elitebook 2560p with :
CPU: Intel(R) Core(TM) i5-2520M (2nd generation)
GPU: Integrated Graphics
WIFI: Intel Corporation Centrino Advanced-N 6205

After several unsuccessful attempts with brunch, I managed to install ChromeOS using a converted ArnoldTheBat Chromium image into ChromeOS using "Croissant" project, but WIFI was not working nor play store (as indicated on the ArnoldTheBat website).

I want to persevere using Brunch and I managed to boot using built image with BIOS/MBR support (UEFI mode is buggy on this laptop and in beta mode in this laptop) but i'm stuck at boot after ROOTFS rebuild.

The following message is printed in loop :

init: shill main process (XXXX) killed by ABRT signal      <== with XXXX = PID
init: shill main process ended, respawning

Thank in advance !

sebanc, 2020-12-13:
Did you try only the latest release with 5.4 kernel or also the previous one with kernel 4.19 ? It seems there are some regressions with the 5.4 kernel.

I think shill process failing is due to a previous error, could you make a short video of the boot process ?

OlivierPonel, 2020-12-13:
Thank for your quick answer.
I tested only with the 5.4 kernel. 
I will try this afternoon making USB Key using the Brunch r86 k4.19 stable 20201014.

OlivierPonel, 2020-12-14:
Hi @sebanc ,

Creating image on USB with the kernel 4.19 worked perfectly !

But after installing ChromeOS on HDD from the USB Key, grub seems to be missing on /dev/sda. 

==> Missing OS message from the BIOS

I can only boot from USB key, but i don't know if I'am on the "live usb" or on the installed chromeOS in the HDD.

The grub ls command (from the USB key) show hd0 and hd1.

sebanc, 2020-12-15:
You would need to setup your bios to boot in EFI mode, it's apparently trying to boot in mbr mode.

Otherwise, you can also create a boot entry manually in your bios pointing to /efi/boot/grubx64.efi or create it using `efibootmgr`.

OlivierPonel, 2020-12-15:
Hi @sebanc ,

Thanks for your answer. Unfortunately i cannot use EFI mode in this BIOS as it is an experimental implementation of it (according to HP an the message displayed when activating this option in the BIOS).

I tried at first to use it but no success => I always get a message saying that there is a probleme with BootOrder.

Unable to change something with efibootmgr, always an error when writing configuration.

I can't permanently save a boot entry, I have to select on boot (in a basic file explorer system) the efi file. That's why i'am using MBR mode :-(.

sebanc, 2020-12-15:
Ah ok I did not remember that you were using the mbr patch...

In that case, the installer from within chromeos will not work, you will have to boot your chromeos flashdrive or a linux live flashdrive, then download the brunch release, the mbr patch and recovery image and use the modified chromeos-install.sh to install it to the hdd.

closed 2026-03-29 by github-actions[bot]