sebanc/brunch #707: Stuck on boot logo

liljackx, 2020-12-13:
Hi, I recently tried to install ChromeOS on my computer but I had lot of problems doing so.
In fact when I install ChromeOS on my external SSD drive with GPT table and UEFI mode and I try to boot from it on my Desktop pc in UEFI mode it get stucked on the boot logo.
I also tried to boot chromeos in debug mode and it works fine... I mean it compiles everything the right way.
I don't know how to check for logs.
Thanks

[image]

sebanc, 2020-12-13:
What's your laptop / pc specs (CPU / GPU) ?

liljackx, 2020-12-13:
> What are your laptop / pc specs (CPU / GPU) ?

I have an Intel Core i5-6600 and a GTX 960 4gb

sebanc, 2020-12-13:
Chromeos seems correctly booted but it expects the display to be plugged on the i915 card. (Nvidia GPU are not supported)

liljackx, 2020-12-13:
> Chromeos seems correctly booted but it expects the display to be plugged on the i915 card. (Nvidia GPU are not supported)

Thank you so much !

closed 2026-03-29 by github-actions[bot]