sebanc/brunch #710: Huge battery drain on sp2017

JakoDel, 2020-12-15:
Title. I've tried installing the latest brunch version (alongside rammus chromeOS 86), but nothing changed :/.
It loses like 10%/hour in standby.

sebanc, 2020-12-15:
AFAIK 10%/hour in normal usage seems more than correct.

Regarding standby, there is an issue in the linux kernel surface devices wifi driver (mwifiex) which makes it very unstable. The only effective patch I found for this prevents the device from being fully suspended in standby but 10% per hour seems very high. I will try to reproduce it as I also have an SP2017.

JakoDel, 2020-12-18:
> AFAIK 10%/hour in normal usage seems more than correct.
> 
> Regarding standby, there is an issue in the linux kernel surface devices wifi driver (mwifiex) which makes it very unstable. The only effective patch I found prevents the device from being fully suspended in standby but 10% per hour seems very high. I will try to reproduce it as I also have an SP2017.

The big standby drain makes using the surface outside quite hard. Hopefully you can reproduce and fix it. Thanks for your amazing work!

bdodroid, 2020-12-19:
I see the standby drain on my Surface go 2. Wonder if theres something specifically with surface devices?

JakoDel, 2020-12-19:
> I see the standby drain on my Surface go 2. Wonder if theres something specifically with surface devices?

just read @sebanc reply

calvinlang, 2020-12-30:
As a temporary workaround is it possible to have the device turn off wifi on suspension/sleep?

JakoDel, 2021-01-08:
> As a temporary workaround is it possible to have the device turn off wifi on suspension/sleep?

I assume the problem here is that sleep "breaks" the driver
Probably @sebanc is busy, but oh well we'll have to wait for him

calvinlang, 2021-01-10:
One thing really interesting I've noticed is that on the Surface Go, if I sleep with the keyboard attached and closed, the drain is about 5% per hour. 

Without the keyboard attached it's about 10% battery drain per hour.

sebanc, 2021-01-17:
To be honest, I did not have the time to look into this issue. The wifi issue I mentioned previously impacts Surface Pro 4 5 & 6, Book 1 & 2 and Laptop 1 I think but other devices notably Surface Go 1 & 2 or more recent Surface devices are not impacted.

Using the latest release in brunch-unstable repo and rammus recovery image, I just suspended my Surface Go for more than 2 hours and the battery dropped by 2% which seems correct. I will try a full overnight suspend.

If you are still interested in this issue, could you install the latest release from brunch-unstable repo and post :
- the recovery image you use
- the drop in battery after a 2 hours suspend 
- same after a 8 hours suspend

calvinlang, 2021-01-18:
Thanks @sebanc! I just installed the unstable release and I will test the battery over today and tomorrow..

calvinlang, 2021-01-19:
I tried out the new brunch-unstable-release. Seems to be the same issue.

Rammus 87
2-hour suspend battery life 92%
8-hour suspend battery life 66%

I understand this isn't a priority. Thank you so much for everything you do.

sebanc, 2021-01-19:
@calvinlang Could you post the same results after adding "suspend_s3" option ?

calvinlang, 2021-01-21:
Absolutely I would be happy to. I'll focus on it this weekend and let you
know.

calvinlang, 2021-01-23:
I still seem to have the exact same battery drain as without the suspend_s3.

Just in case can you let me know if I entered this correctly?

```
rmmod tpm
	img_part=/dev/nvme0n1p6
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path\
		console= vt.global_cursor_default=0 brunch_bootsplash=default suspend_s3
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
```

sebanc, 2021-01-23:
It's not entered correctly, you have to add "options=suspend_s3" instead of just "suspend_s3".

calvinlang, 2021-01-23:
Ahh! Glad I checked I will try it this weekend!

calvinlang, 2021-01-24:
That made a huge difference thank you so much. Battery drain is less than 1% per hour and this is the best tablet I've ever had in my life!!!

JakoDel, 2021-02-01:
> @calvinlang Could you post the same results after adding "suspend_s3" option ? (this is only valid on Surface Go 1/2 and will most likely cause issues on other Surface devices which do not support S3 suspend)

Any fix for the surface pro 2017?