sebanc/brunch #711: No Dualboot Partitions in brunch

mithickartik, 2020-12-15:
My problem is that I want to dual boot windows 10 and chrome OS. But I want chrome OS to have the partitions and not be a img file. My particular laptop is slow in using img files but faster with partitions. So can someone please make a build that can dual boot with partitions meaning windows 10 has its own partition and chrome OS has his own partitions instead of an img file? Thankyou!

mithickartik, 2020-12-17:
@sebanc is there an option?

sebanc, 2020-12-17:
The dual boot image is the best method because:
- ChromeOS wants its specific 12 partitions setup
- Windows does not like chromeos partitions setup

So, the only proper option I can think of would be to use 2 different hdd.

Aside from that, if you want to try it for yourself, you can install chromeos then reduce partition 1 size and install windows using that free space but I can almost guarantee that your install will break shortly. Clearly, if you do this, don't expect support.

mithickartik, 2020-12-19:
Thanks for the help and tips! I might try the steps you wrote but I wont need support because the manufacture of my laptop gave me a recovery drive.

closed 2020-12-19 by mithickartik