sebanc/brunch #713: Brunch free everytime after grub and then success at next boot.

kingo132, 2020-12-16:
After I choose the default boot option in the grub menu, the screen went black, and the system freeze(didn't see the brunch logo).
When I shut down and reboot again, brunch can successfully boot ChromeOS.
It's very regular. One time freeze and next time success, and then next time it will freeze again, and then next time it will succeed again, and so on.
I didn't modify anything. The problem exists when the install is clean.
Does anyone have the same phenomenon that can share some fixes? Or where can I check the log to see what errors have happened?

kingo132, 2020-12-17:
I did some test today, the boot will freeze immediately after grub choosing. The system will hang without any log even in debug mode. I think it may be is a grub problem. Does anyone know how to change another version of grubx86.efi? I replaced it with the one in ubuntu but does not work. Which other versions of grubx86.efi works with brunch?
And one thing really strange, the system will hang only if I click to power off button (chrome does not have a reboot button) in ChromeOS and then it will freeze at the next boot. Maybe it was the power of the mechanism of ChromeOS that cause the problem?

kingo132, 2020-12-17:
I tried to power off or reboot the system from crosh. And here is the test result.
sudo reboot: won't cause boot freeze, it boots well at every time
sudo shutdown -h now: the same as the power off button of ChromeOS, this will cause the boot freeze at the next boot.
So maybe it was the problem of the shutdown mechanism of ChromeOS? Why does it freeze every next boot when shutdown or poweroff?

sebanc, 2020-12-17:
Could you try the new "brunch-unstable" build ?

kingo132, 2020-12-18:
> Could you try the new "brunch-unstable" build ?

OK, I'll try.
I find some clues from ubuntu. Ubuntu will freeze too when reboot from chrome os. Here is the log I grabbed from ubuntu. It trapped into a soft lock, maybe related to some kernel bugs.
I searched the whole internet and tried nearly every fix about this problem, but none is usable. Currently don't know how to work around this. I have to wait for the first boot freeze to reboot and then boot again into chromeos.

[image]

sebanc, 2020-12-18:
In the latest build I reverted this chromiumos kernel specific commit which looks suspicious:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/8257a4bbfec325366af2abb55d7fdd7324f61374

Let me know if it helps

kingo132, 2020-12-18:
> In the latest build I reverted this chromiumos kernel specific commit which looks suspicious:
> https://chromium.googlesource.com/chromiumos/third_party/kernel/+/8257a4bbfec325366af2abb55d7fdd7324f61374
> 
> Let me know if it helps

I tested the r87 unstable, still the same, it will freeze too

sebanc, 2020-12-19:
It seems to be a regression introduced by 5.4 kernel and I have no other idea what it could be related to.

I would suggest that you use kernel 4.19 until I figure this out by replacing "/kernel" in the grub configuration by "/kernel-4.19".

Alternatively, if you install the latest brunch-unstable release, you can try an early build of kernel 5.10 by replacing "/kernel" in the grub configuration by "/kernel-5.10".

sebanc, 2020-12-22:
As a temporary workaround, could you try adding "reboot=pci" or "reboot=acpi" on the kernel command line ?

kingo132, 2020-12-22:
> It seems to be a regression introduced by 5.4 kernel and I have no other idea what it could be related to.
>
> I would suggest that you use kernel 4.19 until I figure this out by replacing "/kernel" in [...]

I tested 4.19. This kernel is too old for my device. It can't boot into the destop.
Then I tested 5.10 with brunch-unstable release. It still freeze but it will output some logs before the freeze. Here is the log.

[image]

kingo132, 2020-12-22:
> As a temporary workaround, could you try adding "reboot=pci" or "reboot=acpi" on the kernel command line ?

These two options will freeze too.
However, when I added "reboot=pci" option to kernel 5.10. It outputs more logs than the screenshot I posted before. From these more logs, I found that it still runs into the kernel soft lockup.

xtibrox, 2022-03-03:
> It seems to be a regression introduced by 5.4 kernel and I have no other idea what it could be related to.
>
> I would suggest that you use kernel 4.19 until I figure this out by replacing "/kernel" in [...]

And how to save after change the kernel to 5.10?