sebanc/brunch #714: Lenovo Thinkpad X1 Carbon 7th gen, no soundcard detected, ChromeOS 87 rammus

umasse, 2020-12-16:
Hello,
I am running ChromeOS 87 with brunch on Thinkpad X1 Carbon 7th generation. Most things work well, but I can't get audio to work. The soundcard was difficult to get it working properly on Ubuntu 20.04, and only on Ubuntu 20.10 is working well.

The lspci output:
`00:1f.3 Audio device [0403]: Intel Corporation Device [8086:9dc8] (rev 11)`
Full lspci here:
[lspci.txt](https://github.com/sebanc/brunch/files/5705419/lspci.txt)

And here is dmesg output:
```
[    1.032630] pci 0000:00:1f.3: [8086:9dc8] type 00 class 0x040380
[    1.032710] pci 0000:00:1f.3: reg 0x10: [mem 0xea23c000-0xea23ffff 64bit]
[    1.032797] pci 0000:00:1f.3: reg 0x20: [mem 0xea000000-0xea0fffff 64bit]
[    1.032949] pci 0000:00:1f.3: PME# supported from D3hot D3cold
[   13.626852] pci 0000:00:1f.3: attach allowed to drvr snd_hda_intel [trusted device]
[   13.626872] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[   13.627027] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   13.629805] pci 0000:00:1f.3: attach allowed to drvr snd_soc_skl [trusted device]
[   13.629825] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[   13.629974] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   13.633799] pci 0000:00:1f.3: attach allowed to drvr sof-audio-pci [trusted device]
[   13.633820] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[   13.633969] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   13.634118] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   13.636266] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   13.644777] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   13.644785] sof-audio-pci 0000:00:1f.3: error: no matching ASoC machine driver found - aborting probe
[   13.644789] sof-audio-pci 0000:00:1f.3: error: failed to get machine info -19
[   13.644923] sof-audio-pci 0000:00:1f.3: error: failed to probe DSP hardware!
```
Full dmesg here:
[dmesg.log](https://github.com/sebanc/brunch/files/5705412/dmesg.log)

Any tips would be welcome!

umasse, 2020-12-17:
For comparison here is the lspci output on Ubuntu 20.10
`00:1f.3 Audio device [0403]: Intel Corporation Cannon Point-LP High Definition Audio Controller [8086:9dc8] (rev 11)`
Full lspci output: 
[lspci-ubuntu.log](https://github.com/sebanc/brunch/files/5707662/lspci-ubuntu.log)

And the dmesg output on Ubuntu 20.10
```
[    0.609642] pci 0000:00:1f.3: [8086:9dc8] type 00 class 0x040380
[    0.609718] pci 0000:00:1f.3: reg 0x10: [mem 0xea23c000-0xea23ffff 64bit]
[    0.609801] pci 0000:00:1f.3: reg 0x20: [mem 0xea000000-0xea0fffff 64bit]
[    0.609948] pci 0000:00:1f.3: PME# supported from D3hot D3cold
[    7.912235] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    7.912238] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    8.078155] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    8.078164] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    8.078361] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    8.078474] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.086288] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    8.136159] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    8.136162] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    8.136166] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    8.236811] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:6:0-18fab
[    8.236813] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:17:0 Kernel ABI 3:16:0
[    8.236814] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    8.237253] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    8.237270] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    8.245520] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:16:0
[    8.245521] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    8.245525] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    8.246817] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    8.246818] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    8.248048] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    8.248050] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    8.248987] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    8.248988] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    8.250063] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    8.250065] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    8.250066] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    8.250067] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    8.250068] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    8.250070] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    8.250898] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    8.250899] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    8.251740] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    8.251741] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    8.252519] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    8.260653] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    8.344625] input: sof-hda-dsp Headset Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[    8.344697] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[    8.344727] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    8.344765] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    8.344796] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    8.344831] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
```

sebanc, 2020-12-17:
Could you try the new "brunch-unstable" build ?

umasse, 2020-12-17:
With the new brunch-unstable build, sound playback works. Microphone does not seem to work, but I have not tried any of the existing microphone debugging options yet.
Thanks for your quick reply!

umasse, 2020-12-20:
I have tried the internal_mic_fix option in grub, but no luck. I will get dmesg and alsa devices details later.

umasse, 2020-12-25:
Output of "dmesg | grep hda":
```
localhost ~ # dmesg | grep hda
[   13.465226] pci 0000:00:1f.3: attach allowed to drvr snd_hda_intel [trusted device]
[   13.465249] snd_hda_intel 0000:00:1f.3: dmic_detect option is deprecated, pass snd-intel-dspcfg.dsp_driver=1 option instead
[   13.468290] snd_hda_intel 0000:00:1f.3: attach allowed to drvr snd_soc_skl [trusted device]
[   13.470263] snd_hda_intel 0000:00:1f.3: attach allowed to drvr sof-audio-pci [trusted device]
[   14.379197] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops intel_dual_link_lvds_callback [i915])
[   14.406069] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[   14.406076] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   14.406081] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   14.406084] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   14.406087] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   14.406090] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
```

amixer -c0 output does not show any internal mic, so the instruction:
```
amixer -c0 set "Internal Mic" toggle
```
does not work. I didn't see anything that I could toggle with any similar name.
I also tried playing with mixer levels with alsamixer -c0, but no change. Internal microphone still doesn't capture anything.

I am attaching full dmesg and amixer outputs.
[amixer.txt](https://github.com/sebanc/brunch/files/5742287/amixer.txt)
[dmesg.log](https://github.com/sebanc/brunch/files/5742288/dmesg.log)

Let me know if I can try anything else, or if anyone knows where else to look for ideas?

umasse, 2020-12-28:
Tried with latest stable. No change with default kernel. No sound card detected with kernel 5.10.

stucki, 2020-12-28:
I have a Thinkpad Carbon X1 7th Gen here but did not try to run brunch on it yet. However, there were issues to make the sound card work with Ubuntu 20.04 so this might help here too:

- Install firmware from https://github.com/thesofproject/sof-bin/tree/stable-v1.6 (clone the repository + run `./go.sh` to install)
- Additionally, blacklist `snd_hda_intel` and `snd_soc_skl` via `/etc/modprobe.d/blacklist.conf`

If one of these is missing, this might explain why sound does not work for you.

stucki, 2020-12-28:
@sebanc Is there any chance to have more commits in this Git repository? I see that https://github.com/sebanc/brunch/tree/r87/firmware-mods/intel/sof-tplg exists but I don't see the details for the change. So maybe it is already up to date, but I can't tell for sure.

Git history would help a lot as it would allow me to submit PRs. We talked about this already in https://github.com/sebanc/brunch/issues/512#issuecomment-688536728, but the situation is still the same...

umasse, 2020-12-28:
@stucki With latest stable, sound output works. I only have problems with the built-in microphone. I don't have an external mic for the headphone jack, so I can't test that.

umasse, 2021-12-18:
Reviving this issue because one year later, mic still doesn't work. I have tried kernel 5.10, and it seems to work well besides the known microphone issue.
Tried disabling Intel HDA using the options, no difference.