sebanc/brunch #715: What makes ChromeOS lagging

danhu2203, 2020-12-17:
Hello

I have installed Chrome OS on my pc Hp Pro,Intel Core I3 2100 and 2gb ram.

I use ChromeOs r80 with brunch release r81

But why my pc so lag

Does  it the firmware problem,brunch release problem,ram problem or hdd problem

ameyjoshi3068, 2020-12-17:
Hi,
It is known that crostini (Chrome OS Linux mode) and `mount internal drives` option causes Chrome OS to lag.
If this is not turned on, then your ram must be the real cause of it.

danhu2203, 2020-12-17:
I didn't turn on any Linux Mode on my Chrome OS but i saw Mount Internal
Drives when booting

How to turn off the Mount internal drives

>

sebanc, 2020-12-17:
I would say the issue is mostly ram and storage.

You can try adding "enforce_hyperthreading=1" on the kernel command line but not sure it will help much.

danhu2203, 2020-12-18:
Ya,Maybe I have to small RAM or old HDD.I will try to solve the problem by
adding Ram or buying SSD.

Btw Thank You for yr advice

danhu2203, 2020-12-18:
👍

closed 2020-12-18 by danhu2203