sebanc/brunch #717: Play Store not working on 10th gen, Possible fix

mithickartik, 2020-12-19:
@sebanc Play store is working on some 10th gen devices and not all. So maybe its a kernel problem, can you plz try to upgrade to kernel 5.6 maybe or 5.10 from 5.4? I am also happy to help.

sebanc, 2020-12-19:
Brunch uses chromiumos kernel which is only available for "long-term" kernel branches (4.19, 5.4 and 5.10).

You can try a first build of 5.10 kernel by installing the latest brunch-unstable release and changing "/kernel" by "/kernel-5.10" in your grub configuration.

I don't think that it will solve this issue but it would be a good surprise if it does.

mithickartik, 2020-12-19:
@sebanc where can I find this brunch-unstable though?

sebanc, 2020-12-19:
It's one of my github repos

mithickartik, 2020-12-19:
oh ok thanks I will try it.

closed 2020-12-19 by mithickartik

princepjamie06832, 2021-03-27:
Check it out at #983