sebanc/brunch #718: Archer T3U USB wireless adapter not working .

iosifc7, 2020-12-19:
Hello ,

I've recently managed to install rammus os 87 version on my old asus laptop . Because the native wireless card of my laptop is only 2.4 GHz i've bought an USB wireless adapter from TP-LInk , model Archer T3U , but the Chrome OS is not recognizing it . Do you have a solution or there is a way to install a driver compatible with this dongle ? 

Thank you , 
Joseph.

sebanc, 2020-12-19:
Hi, this USB wireless adapter needs a specific driver that is not yet implemented in brunch, I can probably implement it but it will take some time.

I will update this issue when it is implemented.

iosifc7, 2020-12-19:
Understood. Thought that maybe there was a way to install it. As I can see the Linux terminal can recognize it but in Chrome terminal is not seeing it.

sebanc, 2020-12-25:
Could you try the latest brunch-unstable with "options=rtl88x2bu" added on the kernel command line ?

jwalcek, 2021-01-08:
@iosifc7 did it ever work?

iosifc7, 2021-03-20:
Sorry for the late . The notifications were in spam section which I have not checked lately so I'll reply now . Unfortunately I haven.t tried again so I decided to get back my laptop to Windows 10 and return the adapter back to the store. Sorry guys .

closed 2026-03-26 by github-actions[bot]