sebanc/brunch #719: Crashes right after boot when using Intel 7260 WLAN adapter with iwlwifi_backport option

francis0525, 2020-12-19:
Hello. I installed Brunch Framework r87 and Chrome OS 87 to my Samsung laptop. I am using Intel 7260 WLAN adapter(mini pci-e type) with 51 pin marked because I am using Sandy Bridge CPU. The WLAN is connected for about 10 secs after boot, but becomes unusable after that. So I applied iwlwifi_backport in options, but the system crashes right after I log in. Intel 7260 is working well in Windows since I dual booted Chrome OS with Windows. I will really appreciate your helps. Below is the spec of my laptop.

Model Name: NT900X3A-A51
CPU: I5-2537M
M/B Chipset: Intel HM65
RAM: DDR3 1333 4GB * 2
SSD: Samsung mSATA SSD 500GB

sebanc, 2020-12-19:
Hi, try to remove "options=iwlwifi_backport" and to add "iwlwifi.11n_disable=1" on the kernel command line

francis0525, 2020-12-19:
I'd first like to thank you for your help. I tried what you said. I don't understand what you mean so I tried two combinations. "~~ enforce_hyperthreading=1 iwlwifi.11n_disable=1 options=acpi_power_button loop.max_ part=16 ~~" and "enforce_hyperthreading=1 options=acpi_power_button loop.max_ part=16,iwlwifi.11n_disable=1"
But either of these combinations sends me back to GRUB again after a few seconds I enter Brunch.

francis0525, 2020-12-19:
This is a full grub data for Chrome OS

menuentry "ChromeOS (boot from disk image)" {
	rmmod tpm
	img_part=/dev/sda5
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug iwlwifi.11n_disable=1 enforce_hyperthreading=1 options=acpi_power_button loop.max_ part=16 img_part=$img_part img_path=$img_path \
		console= vt.global_cursor_default=0 brunch_bootsplash=default
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

sebanc, 2020-12-19:
What do you mean by "I land back to GRUB after a few seconds I enter Brunch" ? (Do you see the chromeos logo ?)

francis0525, 2020-12-19:
No. I only saw the "Loading Brunch Framework" and the came back to the GRUB

sebanc, 2020-12-19:
There is probably an issue with your grub config, try this one:

```
menuentry "ChromeOS (boot from disk image)" {
rmmod tpm
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \
iwlwifi.11n_disable=1 enforce_hyperthreading=1 options=acpi_power_button \
loop.max_ part=16 img_part=$img_part img_path=$img_path console= vt.global_cursor_default=0 brunch_bootsplash=default
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

francis0525, 2020-12-19:
I tried you config, but  I am still having the symptom. The symptom is recorded in the video below

https://user-images.githubusercontent.com/69278848/102684152-936de300-4219-11eb-927e-c8eba295b2fe.mp4

 the same symptom.

sebanc, 2020-12-19:
This issue is most probably not related to the wifi one, try the below config and post another video:
```
menuentry "ChromeOS (boot from disk image)" {
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \
iwlwifi.11n_disable=1 enforce_hyperthreading=1 options=acpi_power_button \
loop.max_ part=16 img_part=$img_part img_path=$img_path
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

sebanc, 2020-12-19:
wait I think I got it, try this config:
```
menuentry "ChromeOS (boot from disk image)" {
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \
iwlwifi.11n_disable=1 enforce_hyperthreading=1 options=acpi_power_button \
loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

francis0525, 2020-12-19:
1. The boot problem is fixed thanks to your config data.
2. The WiFi problem mentioned before appeared again
3. The WiFi is connected for a few seconds. Then it is suddenly disconnected. And finally, the Wifi disappears from the setting(See the picture below)
4. I found that iwlwifi has some problems when using BT and WiFi the same time, and I am forced to use them the same time because the 51 pin is taped for force enabling BT in Sandy Bridge CPU.
5. Do I have to change the WLAN card?

[image]

sebanc, 2020-12-19:
How about this config ?

```
menuentry "ChromeOS (boot from disk image)" {
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \ enforce_hyperthreading=1 options=iwlwifi_backport,acpi_power_button \
loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

francis0525, 2020-12-19:
I applied. The same symptom still appears, and it even turns itself off right after the WiFi setting disappears.

sebanc, 2020-12-19:
Weird, for now I did not have much issues with the intel wifi drivers, could you use the below config and post a dmesg after the wifi fails (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and post dmesg.txt from your downloads folder)

```
menuentry "ChromeOS (boot from disk image)" {
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \
enforce_hyperthreading=1 options=acpi_power_button loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

francis0525, 2020-12-19:
I tried to run the command, but failed. And I found a bunch of error in boot error. I will upload the picture of both
[image]
[image]

sebanc, 2020-12-19:
Those errors are harmless.

After opening crosh (CTRL+ALT+T, type "shell" first, then the command I provided)

francis0525, 2020-12-19:
I see. This is the log data you requested.

[dmesg.txt](https://github.com/sebanc/brunch/files/5719117/dmesg.txt)

sebanc, 2020-12-19:
It seems to be related to this kernel bug which has apparently never been fixed:
https://bugzilla.kernel.org/show_bug.cgi?id=198645

The most viable solution seems to be to disable 80Ghz channel band on router:
```
I found that anyone use Intel Corporation Dual Band Wireless-AC 7260 will crash module with 802.11ac with 80Ghz channel ban. If disable 80Ghz channel band on router, it works fine with 20-40Ghz channel band.
```

francis0525, 2020-12-19:
What it 80Ghz? Do you mean 80Mhz?

francis0525, 2020-12-19:
This is my WiFi router setting. it is 2.4Ghz 40Mhz.

[image]

sebanc, 2020-12-19:
No idea, I just copied the kernel bug content, I don't know much about wireless specs.

What I would suggest, update to the latest "brunch-unstable" repo release.

Then try to change "/kernel" in your grub config with "/kernel-4.19" or "/kernel-5.10"

francis0525, 2020-12-19:
Okay, I will try