sebanc/brunch #720: Eraser on the Surface Pen cap doesn't work properly on surface go (r86/r87, k5.4)

dieson2000, 2020-12-19:
the cap has been recognized as a nib, but it works well as a eraser with k4.19(86/r87)
"options=ipts" seems no improvement

surface go gen1(8/128), r86 and now r87, dual boot with win10, with framework options: "suspend_s3" "advanced_als"

and thanks for your program, now i use this tablet mostly with ChromeOS, it saves this poor touch screen experience

sebanc, 2020-12-25:
Could you try the brunch-unstable release ?

dieson2000, 2020-12-25:
> Could you try the brunch-unstable release ?

did you mean [this](https://github.com/sebanc/brunch-unstable/releases/tag/r87-unstable-20201225)? i will try it.

dieson2000, 2020-12-30:
updated to r87(20201227)，removed  "/kernel-4.19" , now the PEN works well

closed 2021-02-28 by dieson2000