sebanc/brunch #722: PSA: Ryzen Support Here w/ Some Issues

xeu100, 2020-12-19:
Hi everyone! This morning I was able to use the new Zork recovery image to get Ryzen to boot. A lot of experimentation is happening in the Discord, but here's what I've found so far:

Lenovo IdeaPad Flex 14-API
APU: Ryzen 5 3500U
Recovery: zork/morphius
Everything is functional except the following:
* Bluetooth
* Crostini
* Fingerprint Sensor
Notes:
Must use kernel-4.19 in grub options, random shutdowns do happen, however it seems to have calmed down. Appears bluetooth cannot be used without wifi being shut off, requires newer kernel. Crostini failure likely related to IOMMU issues, causes a segfault in dmesg

Make sure to use the 4.19 kernel version. Here's the direct link to the zork recovery image. https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin.zip

wggglggg, 2020-12-20:
> Hi everyone! This morning I was able to use the new Zork recovery image to get Ryzen to boot. A lot of experimentation is happening in the Discord, but here's what I've found so far:
>
> Lenovo IdeaPad [...]

Brunch r86 k4.19 stable 20201014 + chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin + R5 APU 3400G

didn`t work. 
can be installed,  but nothing show up on the screen,

krtn0828, 2020-12-20:
I also tested it with a RyzenAPU.

APU: Ryzen 3 3200G
(GPU: Radeon vega 8)
M/B: ASUS PRIME A320M-K
Recovery: Zork
Brunch: R86 k4.19

If you install directly to disk using the Linux Terminal, you won't even see GRUB, so run the following command
`sudo bash chromeos-install.sh -src <recovery image> -dst chromeos.img`
Then I installed it to disk with Rufus.
This time GRUB showed up, but the Brunch log blacked out halfway through and didn't work after that.

xeu100, 2020-12-20:
> Brunch r86 k4.19 stable 20201014 + chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin + R5 APU 3400G
>
> didn`t work.
> can be installed, but nothing show up on the screen,

> I also tested it with a RyzenAPU.
>
> APU: Ryzen 3 3200G
> (GPU: Radeon vega 8)
> M/B: ASUS PRIME A320M-K
> Recovery: Zork
> Brunch: R86 k4.19
>
> If you install directly to disk using the Linux Terminal, you [...]

Hm, could you guys use the ChromeOS debug mode, use brunch 87 w/ 4.19 flags, and use Chrome Recovery to flash your USBs? This would better match my setup.

sar777-dev, 2020-12-20:
Maybe this method only supports to mobile(laptop) amd ryzen apu(s) not desktop apu(s) for now.

wggglggg, 2020-12-20:
> Hm, could you guys use the ChromeOS debug mode, use brunch 87 w/ 4.19 flags, and use Chrome Recovery to flash your USBs? This would better match my setup.

Brunch 87 w/ 4.19 flags?, you mean this one unstable https://github.com/sebanc/brunch-unstable/releases/download/r87-unstable-20201217/brunch_r87_unstable_20201217.tar.gz ? and where is the Chromes OS debug mode? Can you record and put it to Youtube?

xeu100, 2020-12-20:
I mean 4.19 with Brunch 87, you can check the releases and it tells you how to use kernel 4.19. Debug mode is an option in the latest brunch's GRUB.

wggglggg, 2020-12-21:
> I mean 4.19 with Brunch 87, you can check the releases and it tells you how to use kernel 4.19. Debug mode is an option in the latest brunch's GRUB.

replacing  likie this ? 

https://ibb.co/R0vjYmR
and i install to Vmware, and found something ,  the process only going to here ,and seems stop to move anymore, i waited long time for wanting to see what will happen, the boot just stopped there.
https://ibb.co/Fz4p84M

wggglggg, 2020-12-21:
> Hm, could you guys use the ChromeOS debug mode, use brunch 87 w/ 4.19 flags, and use Chrome Recovery to flash your USBs? This would better match my setup.

I used debug mode,  Chrome logo showed up 1 sec on the , then and the screen turned black,  I could see the screen button light. 
 Use the 4.19 kernel   like this?  https://ibb.co/R0vjYmR
Use Chrome Recovery to flash the USB......  this App?     https://ibb.co/xX5vhF9  I`m working on it , will fix it not sure.

krtn0828, 2020-12-21:
> Hm, could you guys use the ChromeOS debug mode, use brunch 87 w/ 4.19 flags, and use Chrome Recovery to flash your USBs? This would better match my setup.

Hmm...
Using the Chrome Recovery Utility, I can't proceed from here at all...
[image]

xeu100, 2020-12-21:
> replacing likie this ?
>
> https://ibb.co/R0vjYmR
> and i install to Vmware, and found something , the process only going to here ,and seems stop to move anymore, i waited long time for wanting to see [...]

Yes, but VMWare doesn't work, you need to install it bare metal.

wggglggg, 2020-12-22:
> Yes, but VMWare doesn't work, you need to install it bare metal.

failed again [https://ibb.co/RNvFCVc](url)

xeu100, 2020-12-22:
> > > > I mean 4.19 with Brunch 87, you can check the releases and it tells you how to use kernel 4.19. Debug mode is an option in the latest brunch's GRUB.
> > > 
> > > 
> > > replacing likie this ?
> > > https://ibb.co/R0vjYmR
> > > and i install to Vmware, and found something , the process only going to here ,and seems stop to move anymore, i waited long time for wanting to see what will happen, the boot just stopped there.
> > > https://ibb.co/Fz4p84M
> > 
> > 
> > Yes, but VMWare doesn't work, you need to install it bare metal.
> 
> failed again, 20minutes later , still like this https://ibb.co/RNvFCVc

Definitely looks like a lack of support for your cpu's graphics. See if the Ryzen 5 C-series has a cpu that's near-identical to yours, if there's one then it'll be possible in the future.

krtn0828, 2020-12-22:
> > > 皆さんこんにちは！今朝、新しいZorkリカバリイメージを使用してRyzenを起動することができました。Discordでは多くの実験が行われていますが、これまでに見つけた
> > > ものは次のとおりです
> > > 。LenovoIdeaPadFlex 
> > > 14-API 
> > > APU：Ryzen 5 3500Uリカバリ：zork / morphius以下を除いてすべてが機能しています。
> > > 
> > > * ブルートゥース
> > > * クロスティーニ
> > > * 指紋センサー
> > >   注：
> > >   grubオプションでkernel-4.19を使用する必要があります。ランダムなシャットダウンが発生しますが、落ち着いたようです。Bluetoothをオフにしないと使用できないようです。新しいカーネルが必要です。IOMMUの問題に関連している可能性が高いCrostiniの障害は、dmesgでセグメンテーション違反を引き起こします
> > > 
> > > 必ず4.19カーネルバージョンを使用してください。これは、zorkリカバリイメージへの直接リンクです。https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin.zip
> > 
> > 
> > ブランチr86k4.19安定版20201014+ chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin + R5 APU 3400G
> > 動作しませんでした。
> > インストールできますが、画面に何も表示されません。
> 
> > また、RyzenAPUでテストしました。
> > APU：Ryzen 3 3200G 
> > （GPU：Radeon vega 8）
> > M / B：ASUS PRIME A320M-K
> > リカバリ：Zork
> > ブランチ：R86 k4.19
> > Linuxターミナルを使用してディスクに直接インストールすると、GRUBも表示されないため、次のコマンドを実行します。
> > `sudo bash chromeos-install.sh -src <recovery image> -dst chromeos.img`
> > 次に、Rufusを使用してディスクにインストールしました。
> > 今回はGRUBが表示されましたが、ブランチログが途中で真っ暗になり、その後は機能しませんでした。
> 
> うーん、ChromeOSデバッグモードを使用し、4.19フラグ付きのブランチ87を使用し、Chromeリカバリを使用してUSBをフラッシュできますか？これは私の設定によく一致します。

Hmm ... 
I tried it and GRUB and Brunch started, but I didn't even see the Chrome OS logo.  
I think there is a problem with desktop APU graphics, or APU graphics with Zen, Zen + architecture

xeu100, 2020-12-22:
> Hmm ...
> I tried it and GRUB and Brunch started, but I didn't even see the Chrome OS logo.
> I think there is a problem with desktop APU graphics, or APU graphics with Zen, Zen + architecture

Did you run it in Debug mode with kernel 4.19 on the latest brunch? If so, could you give me more info. It should be at least showing some input.

sebanc, 2020-12-25:
I tried to improve ryzen support in the latest brunch-unstable release, could you try it ?

wggglggg, 2020-12-25:
waiting for it

wggglggg, 2020-12-25:
few days ago ,  installing stopped there    https://ibb.co/RNvFCVc

1 hour ago, tried again with unstable 20201225 Brunch  bare metal.
seems you handed the old problem, but have a new one,  It`s blacked
https://ibb.co/8MjRbK6
https://ibb.co/5TV8xD5
https://ibb.co/thqvvQ0

 
Brunch r87 unstable 20201225 + chromeos_13505.73.0_zork_recovery_stable-channel_mp-v4.bin + R5 APU 3400G

wggglggg, 2020-12-25:
> I tried to improve ryzen support in the latest brunch-unstable release, could you try it ?

xeu100, 2020-12-27:
Crostini soon to be functional in a brunch-unstable build, the issue was found.

wggglggg, 2020-12-27:
> Crostini soon to be functional in a brunch-unstable build, the issue was found.

I`m downloading  it now.

xeu100, 2020-12-27:
> > Crostini soon to be functional in a brunch-unstable build, the issue was found.
> 
> I`m downloading it now.

Oh! Wait, not that one. I'll upload it when it's done building. Sorry, I meant a future one.

wggglggg, 2020-12-27:
> > > Crostini soon to be functional in a brunch-unstable build, the issue was found.
> > 
> > 
> > I`m downloading it now.
> 
> Oh! Wait, not that one. I'll upload it when it's done building. Sorry, I meant a future one.

OK, got it  Thanks

krtn0828, 2020-12-27:
> > うーん...
> > 試してみて、GRUBとブランチが始まりましたが、ChromeOSのロゴすら見えませんでした。
> > デスクトップAPUグラフィックス、またはZen、Zen +アーキテクチャのAPUグラフィックスに問題があると思います
> 
> 最新のブランチでカーネル4.19を使用してデバッグモードで実行しましたか？もしそうなら、私にもっと情報を教えてもらえますか。少なくとも何らかの入力を示している必要があります。

> Did you run it in Debug mode with kernel 4.19 on the latest brunch? If so, could you give me more info. It should be at least showing some input.

Sorry I'm late.  The branch will be blacked out after the last photo is displayed.  The computer itself is running, so I think it's probably a graphics issue.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

xeu100, 2020-12-27:
It doesn't have the usual Frecon Freon Master error, meaning that AMDGPU is there, very odd.

xeu100, 2020-12-27:
https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing

That's my build, not sebanc's. It'll work but I recommend moving to his later on.

wggglggg, 2020-12-27:
> https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing
> 
> That's my build, not sebanc's. It'll work but I recommend moving to his later on.

downloading

krtn0828, 2020-12-27:
> https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing
> 
> That's my build, not sebanc's. It'll work but I recommend moving to his later on.

Thank you, I'm downloading

wggglggg, 2020-12-27:
> > https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing
> > That's my build, not sebanc's. It'll work but I recommend moving to his later on.
> 
> Thank you, I'm downloading

How`s your installing of  the latest one ?  for me , in Vmware , blacked again,   replace /kernal-4.19 and Debug mode  still can`t see the Chrome OS desk.  Going to try bare-metal

wggglggg, 2020-12-27:
blacked , 'No Signal' is on the Display

发自我的小米手机

https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing

krtn0828, 2020-12-27:
> How`s your installing of the latest one ? for me , in Vmware , blacked again, replace /kernal-4.19 and Debug mode still can`t see the Chrome OS desk. Going to try bare-metal

I'm trying Xeu100's build and if that doesn't work, I'm thinking of using the newest Sebanc's unstable build :)

wggglggg, 2020-12-27:
OK, go on .

发自我的小米手机

https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing

wggglggg, 2020-12-27:
> https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing
> 
> That's my build, not sebanc's. It'll work but I recommend moving to his later on.

Didn`t handle The issue

xeu100, 2020-12-27:
As I said, it's not possible to run on virtualization. Bare metal would be your best shot, if it still doesn't work then there might not be a solution. Try the new unstable but there are limited options moving ahead

wggglggg, 2020-12-27:
> 
> As I said, it's not possible to run on virtualization. Bare metal would be your best shot, if it still doesn't work then there might not be a solution. Try the new unstable but there are limited options moving ahead

thank you ,  will try the unstable new one tomorrow.

cngzcelik, 2020-12-27:
> https://drive.google.com/file/d/1-CRNUwnDf9ou6EVZNPVDVSrx-nH8WIw9/view?usp=sharing
> 
> Bu benim yapım, Sebanc'ın değil. İşe yarayacak ama daha sonra ona geçmenizi öneririm.

Günlerdir bu hatayı çözmek için uğraşıyorum ve şimdi senin yaptığın Amd Brunch ile deneyeceğim. Teşekkürler..

xeu100, 2020-12-27:
I'm closing this issue now as the latest brunch-unstable has full kernel support for every feature! (Even Crostini!)

Just wanted to let people know, but everyone with their own issues should totally feel free to open new ones and tag me, I'll do what I can and experiment with my brunchbook.

closed 2020-12-27 by xeu100

cngzcelik, 2020-12-27:
> I'm closing this issue now as the latest brunch-unstable has full kernel support for every feature! (Even Crostini!)
> 
> Just wanted to let people know, but everyone with their own issues should totally feel free to open new ones and tag me, I'll do what I can and experiment with my brunchbook.

> 

> 
[image] @xeu100 xeu100 Sayende ChromeOS'u Ryzen 3 cihazımda çalıştırdım. Günlerdir (Fb0: switching to amdgpudrmfb from EFI VGA) sorunu ile uğraşıyordum ve sonunda çalıştı. Sadece Wifi çalışmıyor. Tekrar teşekkürler.

krtn0828, 2020-12-30:
[image]
I use latest brunch 5.10 kernel,  and its working!!
thanks for many support!!

wggglggg, 2020-12-30:
> [image]
> I use latest brunch 5.10 kernel, and its working!!
> thanks for many support!!

me too

xeu100, 2020-12-30:
Great! If you guys could please log your device hardware with what is and is not working either here or on the Discord I'll make sure to keep note. Things like CPU/GPU, device type, anything that is or is not working, etc.

krtn0828, 2020-12-30:
> Great! If you guys could please log your device hardware with what is and is not working either here or on the Discord I'll make sure to keep note. Things like CPU/GPU, device type, anything that is or is not working, etc.

All right,I will log on the Discord soon!

wggglggg, 2020-12-31:
> Great! If you guys could please log your device hardware with what is and is not working either here or on the Discord I'll make sure to keep note. Things like CPU/GPU, device type, anything that is or is not working, etc.

CPU:  Ryzen 5   3400G (APU)
GPU:  Radeon RX Vega 11 Graphics
MAINBOARD: ASRock DeskMini X300 Barebon (is a minibox include Mainboard)
Memory:  Crucial 8GB DDR4 3200MHz - CT8G4SFS832A- 260-Pin SO-DIMM
Chipset:  AMD X300
Graphics Output:  DP
WLAN:  AX200
SSD: HP 500G SSD M.2(NVMe协议) EX900
Keyboard: Ducky DK2108S
Sound , 4K output  no problem, WiFi 5G  is OK,  Google App Store is working well 
but I haven`t tried Bluetooth , and the Num buttons (on the right of the Keyboard, is not working well