sebanc/brunch #723: MacBook Pro 8,2 Compatibility Report (early 2011)

vervallsweg, 2020-12-19:
- [x] Does the Android Subsystem (Android apps) run?
 - [x] Is Linux container install option available?
 - [x] Does Linux container install?
 - [x] Does Linux terminal run?
 - [x] Is Bluetooth working?
 - [x] Is WiFi working?
 - [ ] Is the camera working? (info below)
 - [x] Is microphone working?
 - [ ] ~~Does the touchscreen (if your device has one) work~~
 - [x] Does trackpad and gestures work?

Well this one took me a while. It boots brunch fine and just gets stuck at the chromeos logo (#688). I suspected a graphics issue with it's amd radeon chipset. I tried the usual thing of `nomodeset` and `disable noveau`. The latter of which, I think is nvidia specific, but regardless, non worked and wouldn't even boot to the chrome logo. 

### Getting brunch to boot

So I choose the nuclear option and disabled the amd graphics [entirely in EFI](https://apple.stackexchange.com/questions/166876/macbook-pro-how-to-disable-discrete-gpu-permanently-from-efi). You can either boot into recovery, or use a live distro to set the efivars.

After that brunch boots to the setup screen 🎉 
Grub is still broken however, which makes this super weird. You have to set grub to [vga mode](https://forums.linuxmint.com/viewtopic.php?t=308344) `GRUB_VIDEO_BACKEND=vga
` . Idk if that is the actual fix for it, at least this made it work for me. If you don't do this, your screen will just freeze as long as grub is displaying it's UI and you obviously cannot set any options.

### What is not working

#### display/keyboard back-light control
Dunno, don't care, didn't investigate. Works in Ubuntu out of the box though.

#### camera
Doesn't work and regardless of image version, `dmesg` displays this error message #691 . I tried downgrading, upgrading. Error stayed the same, camera still didn't work.

I suspect a driver issue. However, on Ubuntu it works out of the box and seemingly uses a standard uvc driver. I read up on the `isight-firmware-tools` package. However I don't think Ubuntu is using it out of the box either. I gave up after that.

If anyone finds some smart solution for the webcam that would be awesome. For now, I'll stick to Ubuntu.

Brax84, 2021-02-16:
Macbook 8,2 15” 2011

Everything works except key brightness on keyboard solve via command on shell
Facetime camera working after this step [#552](https://github.com/sebanc/brunch/issues/558#issuecomment-702829290) 
For varius tweak [#883](https://github.com/sebanc/brunch/issues/883)

zampemo, 2021-04-07:
> Macbook 8,2 15” 2011 with rammus r88
> 
> Everything works except key brightness on keyboard solved via command on shell
> Facetime camera working after this step [#552](https://github.com/sebanc/brunch/issues/558#issuecomment-702829290)
> For varius tweak [#883](https://github.com/sebanc/brunch/issues/883)

Hi Brax, I tried to make my camera work with those steps but it's a no-go for my MacBook Air 2013 unfortunately.

closed 2026-03-26 by github-actions[bot]