sebanc/brunch #725: [Guide] How I fixed my iio ACPI Ambient Light Sensor (ALS)

jcarrz1, 2020-12-20:
##Open Shell

Ctrl + alt + T (open shell)
type `shell`

##Diagnose

The fix only applies to devices with an Ambient Light Sensor using the acpi-als driver in linux. If you have an ALS that uses the acpi-als driver, you'll be able to find it this way: (brunch kernel 5.4)

```
cd /sys/bus/iio/devices/
ls
cd [device that shows up]
ls
```
check for some variables here that say "illuminance", if they're present you can see if they change under different lighting conditions with `cat`

## Apply Fix
```
cd /var/lib/power_manager
sudo nano has_ambient_light_sensor
```
type the number `1`
ctrl+O, ctrl+x

```
sudo mkdir board_specific
cd board_specific
sudo nano has_ambient_light_sensor
```
type the number `1`
ctrl+O, ctrl+x

```
sudo reboot
```

to: sebanc/devs, this can probably be fixed in chromeos-config. 
rammus has 2 ambient light sensors, my laptop had only 1, and this caused my issue. Chrome OS appears to use the first sensor for keyboard backlight and the second sensor for display backlight, which doesn't make sense for laptops that only have one ALS.

https://chromium.googlesource.com/chromiumos/platform2/+/refs/heads/release-R88-13597.B/power_manager/powerd/system/ambient_light_sensor.cc

-James

sebanc, 2020-12-27:
Most devices which have only 1 ambiant light sensor seem to work fine with "has_ambient_light_sensor" set to 2. I am trying to understand of why your device would need this.

Which recovery image do you use ?

jcarrz1, 2020-12-27:
I'm using rammus.

I tried to root-cause by reading the code. It seems as I described above,
It tried to use it for keyboard, there was a TYPE enum somewhere in the
code and the first member of the enum is keyboard backlight