sebanc/brunch #726: Add Realtek RTL88x2U Wifi

Freaklimit, 2020-12-20:
Hi everyone,

is it possible to add or install Wifi driver on Chrome OS. If so how can I do that. I have the official driver package from the manufacturer for Windows, Mac and also Linux, but I don´t know how i can add this. On other Linux distribution it works with this driver.

Thank you in advance for the help.

sebanc, 2020-12-25:
I have added the rtl88x2bu module in the latest brunch-unstable build (use "options=rtl88x2bu"), could you try it ?

Freaklimit, 2020-12-28:
I updated the brunch and added the option in grub, but it doesn't worked for me. I have a wifi dongle from ASUS (ASUS USB-AC53 Nano). I got the driver from the manufacturer page [https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-AC53-Nano/HelpDesk_Download/] the EU version. I tried it on Linux Mint 20, Linux Buddgie and Ubuntu 20.04.1 LTS and it worked there.

sebanc, 2020-12-31:
Could you attach the output of "lsusb" in crosh shell ?

Could you also post a dmesg with "options=rtl88x2bu" enabled ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder)

Freaklimit, 2021-01-03:
That's what "lsusb" puts out:

`chronos@localhost / $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0b05:184c ASUSTek Computer, Inc. 
Bus 001 Device 003: ID 1c4f:0063 SiGma Micro 
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

And here the dmesg attached file
[dmesg.txt](https://github.com/sebanc/brunch/files/5761622/dmesg.txt)

sebanc, 2021-01-03:
Your kernel command line is :`Command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug console= vt.global_cursor_default=0 brunch_bootsplash=default`

"options=rtl88x2bu" has not been added. Could you add it and see if it works ?

Freaklimit, 2021-02-02:
Hello again,

excuse me for this very late answer, but i want to tell, that it works now with the new brunch r88 stable 20210131 and the Chrome OS Version 88.0.4324.109. I don't know if the problem is gone with the brunch update or the OS update. I've done it both at the same time. If someone have the same problem update the version above and add the option "rtl88x2bu" to your grub config.

Thank you very much for your fast help and for this great project.

closed 2021-02-02 by Freaklimit