sebanc/brunch #729: Mic and Camera not working - Dell XPS 13 9300

fabiofeichtinger, 2020-12-20:
Hello. 

I am using brunch framework with Dell XPS 13 (2020) 4K, touchscreen with Chrome OS 87 **rammus**

I can`t get to work my bult-in camera and my bult-in microphone, and I use this hardware on a daily base with other Linux distro`s.

Thanks

heliofreitasbr01, 2020-12-23:
I have the same problem with a XPS 14 and a Macbook Pro 2012 using Sammus. No cam and no mic after Chrome OS 86.

Lucaacer, 2020-12-28:
I have just faced the same issue with sammus and Chrome OS 86.

Lenovo thinkpad t430 and mediacom flexbook edge 13.

Camera of both devices used to work with older releases.

Updating the framework did not solve it.

Thanks.

Lucaacer, 2020-12-28:
I moved to the beta channel and this issue is still there in chrome OS 88.

sebanc, 2020-12-28:
The only recent change in brunch is the switch to 5.4 kernel, you can try to switch back to 4.19 kernel by replacing "/kernel" in the grub configuration with "/kernel-4.19"

Lucaacer, 2020-12-30:
I have a working live usb with chrome 86 and it's fine with 

crosh> uname -r
5.4.75-brunch-sebanc

So it's not your kernel, it's something related to the update.

I own a real chromebook too and camera still works with chrome 87, though.

Let me know if you need any log. While using the live usb, this is what I get

crosh> shell
chronos@localhost / $ lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 006: ID 04f2:b2db Chicony Electronics Co., Ltd 
Bus 001 Device 005: ID 0a5c:21e6 Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad]
Bus 001 Device 004: ID 147e:2020 Upek TouchChip Fingerprint Coprocessor (WBF advanced mode)
Bus 001 Device 003: ID 0781:5583 SanDisk Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

And this is the camera 

Bus 001 Device 006: ID 04f2:b2db Chicony Electronics Co., Ltd 

I will check the other install.

Lucaacer, 2020-12-30:
This one could be of some help

chronos@localhost / $ dmesg | grep camera
[   19.091364] init: cros-camera main process ended, respawning
[   24.348696] init: cros-camera main process ended, respawning
[   29.484403] init: cros-camera main process ended, respawning
[   34.578693] init: cros-camera main process ended, respawning
[   39.887958] init: cros-camera main process ended, respawning
[   45.359270] init: cros-camera main process ended, respawning
[   50.597719] init: cros-camera main process ended, respawning
[   55.946200] init: cros-camera main process ended, respawning
[   61.248786] init: cros-camera main process ended, respawning
[   66.748931] init: cros-camera main process ended, respawning
[   71.938734] init: cros-camera main process ended, respawning
[   77.090760] init: cros-camera main process ended, respawning
[   82.228722] init: cros-camera main process ended, respawning
[   87.370741] init: cros-camera main process ended, respawning
[   92.585862] init: cros-camera main process ended, respawning
[   97.699767] init: cros-camera main process ended, respawning
[  102.917230] init: cros-camera main process ended, respawning
[  108.514898] init: cros-camera main process ended, respawning
[  113.799914] init: cros-camera respawning too fast, stopped

heliofreitasbr01, 2020-12-30:
The only way to both machines working fine is a new installation with any kernel (5.4 or 4.19) and Sammus Recovery 86. With any other OS version, channel or update it will not working anymore.

sebanc, 2020-12-30:
Chromeos r87 broke camera on actual chromebooks too. For now I don't really have a solution.

Actually I forgot about it but you can try the solution in #627.

heliofreitasbr01, 2020-12-30:
I did it too. But with no effect. Well, let`s wait until we get a solution. Thanks.

Lucaacer, 2021-01-03:
> Chromeos r87 broke camera on actual chromebooks too. For now I don't really have a solution.
> 
> Actually I forgot about it but you can try the solution in #627.

I have just checked and on my daughter's chromebook, running Chrome OS 87, camera still works.

The build is Pyro.

Regarding my 2 devices, camera does not work even after moving to beta or dev channel, so Google must have added something which breaks your kernel/settings.

It's not a big issue moving back to chrome OS 86, which is more or less like chrome 87, but it would be a shame if we should stay on that release forever.

So, let me know if I may be of any help with logs and so on.

Thanks as usual.

Btw, upgrading to the latest framewok has just broken support to the touchscreen on my Mediacom Flexbook Edge 13:(

sebanc, 2021-01-03:
Regarding camera, it might be due to the camera config I use as base (from rammus recovery image) not being adapted for your camera. I would need someone to try other configs and see if they improve things, if they do I could add them as framework options.
The camera config is in /etc/camera/camera_characteristics.conf

Regarding touchscreen, I think I know what the issue is, I am working on it.

Lucaacer, 2021-01-03:
Thanks. I replaced the current config file with the older one that I had taken from the live usb and rebooted, but nothing changes.

sebanc, 2021-01-03:
What I meant is that you could try other recovery images /etc/camera/camera_characteristics.conf (brunch always uses its own /etc/camera/camera_characteristics.conf but you can extract a different configuration file from another recovery image and replace the brunch one).

sebanc, 2021-01-03:
Could you try the touchscreen in the new brunch-unstable ?

Lucaacer, 2021-01-04:
> Could you try the touchscreen in the new brunch-unstable ?

Sure, but where can I find the file?

sebanc, 2021-01-04:
In the brunch-unstable repository of my github

Lucaacer, 2021-01-04:
> In the brunch-unstable repository of my github

I tested it on my mediacom and in chrome 86.0.4240.198, after flashing the latest unstable brunch framewoork, camera still works, while the touchscreen still fails.

Anyway, when I first boot windows (which is on another ssd) and then chrome OS, the touchscreen works, which is the behaviour of some months ago.

kkoouu, 2021-02-18:
I'm on XPS 9360 and latest DEV ( Version 90.0.4411.0 (Official Build) dev (64-bit) ) .... camera works but mic doesn't .... I even tried a USB mic, it was detected but not working as well - no sound ...

neerajverma04, 2021-04-29:
I got it working with your command on XPS 9365 but i had to set amixer -c1 set "Internal Mic