sebanc/brunch #733: Fix wacom touchscreen wacf2200 (Thinkpad and lenovo)

Alexfebbraio, 2020-12-21:
Hello
Some thinkpad and lenovo laptops use the wacon touchscreen (wacf2200) but there is an error in the linux kernel and it doesn't work.
Fortunately, the fix was found.

https://patchwork.kernel.org/project/linux-acpi/patch/20201211021814.36193-1-hui.wang@canonical.com/

I modified the acpi_pnp.c file and recompiled the kernel on ubuntu, fedora and manjaro and the touchscreen finally works perfectly.
It would be nice to make the change on the 5.4 brunch kernel but I don't know how

wangbingrui, 2020-12-23:
Hi
I have seen everyone work fix the wacf2200 problem. Thank you everyone.
I have a lenvo laptop with wacf2200 and i want install chrome os on it.I search online and found this. https://github.com/kitakar5525/chromeos-kernel-linux-surface#how-to-install
and this
https://github.com/kitakar5525/chromeos-kernel-linux-surface/releases in "4.19.152 and 5.4.72"-->Installation-->for brunch
It mentioned how to install the linux kernel for surface chrome os, I hope it helps. My knowledge of linux is limited, if you succeed, please write a short tutorial. thank you very much.

Alexfebbraio, 2020-12-23:
> Hi
> I have seen everyone work fix the wacf2200 problem. Thank you everyone.
> I have a lenvo laptop with wacf2200 and i want install chrome os on it.I search online and found this. [...]

Thank you. 
I try to understand. I also hope that someone more experienced than me will read the message

sebanc, 2020-12-25:
I have added the fix in the latest brunch-unstable build, could you try it ?

Alexfebbraio, 2020-12-25:
Thanks Sebanc. The touchscreen works perfectly.
I just have to figure out by reading on the internet how to enable tablet mode to bring up the virtual keyboard

sebanc, 2021-01-03:
Hi @Alexfebbraio,

It seems this patch has introduced a regression for some users, I am going to target this patch specifically for Lenovo systems but I would need some more information to do so.

Could you please attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your downloads folder)

Alexfebbraio, 2021-01-03:
> Hi @Alexfebbraio,
>
> It seems this patch has introduced a regression for some users, I am going to target this patch specifically for Lenovo systems but I would need some more information to do so. [...]

Hi. Ok
[dmesg.txt](https://github.com/sebanc/brunch/files/5761908/dmesg.txt)

sebanc, 2021-01-04:
Could you verify that touchscreen still works in the new brunch-unstable ?

Alexfebbraio, 2021-01-04:
> Could you verify that touchscreen still works in the new brunch-unstable build ?

I did the test. The touchscreen works