sebanc/brunch #743: Surface pro 7 Heat

MohamedAlmahmood, 2020-12-24:
Thank you for the amazing work.
Chrome os has been running great on the 4.12 kernel. 
However, I noticed that it is heating my device more than windows. Watching YouTube at 1440p using the android app and downloading a torrent using another android app heats up he cpu to around 55 degrees. It rarely gets this hot on windows. Shouldn't chrome os be lighter on the CPU than windows? Is this normal especially since I'm only running two apps? Is there anything I can do to make the CPU cooler? 
Hyperthreading is turned off. Do you think I should enable it?
Any advice is much appreciated.
Thank you.

MohamedAlmahmood, 2020-12-24:
I think I fixed the issue. I will let you know once I do more testing. But still any advice is appreciated.

ZZZXXX0110, 2021-02-05:
How did you manage to boot on the 4.19 kernel? Do you have the i3 version of SP7?

nadlax, 2021-03-14:
@MohamedAlmahmood how'd you manage to fix it? I'd really love to know

Vinson1014, 2021-03-28:
My surface pro 7(i5 128G) is heat when ruuning ChromeOS either. Can you tell us how you fix it?

nadlax, 2021-03-31:
> My surface pro 7 (i5 128G) is heat when ruuning ChromeOS either. Can you tell us how you fix it?

Remove 'mount_internal_drives' from. If it still doesn't bring down heating issues or if you can't live without access to internal partitions, install Fyde OS. That's what I did. Fyde OS Is better

ijinesh, 2021-11-30:
> I think I fixed the issue. I will let you know once I do more testing. But still any advice is appreciated.

Hi,
Can you share your fix for this overheating issue?

closed 2026-03-25 by github-actions[bot]