sebanc/brunch #744: [Intel Iris Xe 11th Gen] Stuck on boot logo then turns to black

TreeTopSky, 2020-12-25:
I'm using the Rammus r87 build with the latest brunch on the 5.4 kernel. The USB install worked like a charm and booting into the brunch first time setup went well. After it completes the ChromeOS logo displays, hangs, then the display turns off. No backlight, and no read/write on usb stick.

I'm attempting install on the HP Envy x360 tablet/touchscreen laptop.

xeu100, 2020-12-25:
I believe there aren't any 11th gen Chromebooks at this time, so you may be out of luck for now.

sebanc, 2020-12-26:
@TreeTopSky Could you try with brunch unstable and kernel 5.10 ?
(replace /kernel in the grub configuration by /kernel-5.10)

TreeTopSky, 2020-12-26:
> @TreeTopSky Could you try with the brunch-unstable release and kernel 5.10 ?
> (replace /kernel in the grub configuration by /kernel-5.10)

Where can I find the unstable release?

sebanc, 2020-12-26:
Look for the brunch-unstable repo on my github

TreeTopSky, 2020-12-26:
Now it's just hanging, no black screen.

sebanc, 2020-12-26:
When you are stuck on the chromeos logo, can you press CTRL+ALT+F2 to access a console session ?

If so, could you login as "root", then run "dmesg" and attach a photo of the screen ?

TreeTopSky, 2020-12-26:
[image]

sebanc, 2020-12-26:
There is an issue with sound but it is probably not the reason why the system freeze on boot.

What is the output of `dmesg | grep "drm \| i915"` ?

TreeTopSky, 2020-12-26:
[image]

sebanc, 2020-12-28:
Sorry but I don't think 11th gen are supported in chromeos currently, I added a note in the main readme. It might still change in a future chromeos version.

yuhan6665, 2020-12-28:
Thanks for your work. Hope for the future

johnnylau34, 2021-03-07:
hmm, read a tigerlake-based ChromeBook should be coming some time in Q2...let's hope for the best in the coming quarter:
https://www.notebookcheck.net/Asus-launches-ChromeBook-CX9400-and-ChromeBook-Flip-CX5500-premium-grade-laptops-with-Intel-Tiger-Lake-U-CPUs.514857.0.html

skylatian, 2021-04-19:
any update on this? having the same issue on a razer book 13 (i7 xe 11th gen)

johnnylau34, 2021-06-02:
hey not sure if anyone is aware, now it's stated on the frontpage that:

Currently:

"rammus" is the recommended image for devices with 1st gen Intel CPU up to 9th gen.
_"volteer" is the recommended image for devices with Intel 10th and 11th gen CPU (Intel 11th Gen devices need kernel 5.10, read "Changing kernel version" section)._

I would definitely go and check it out on my x1 titanium yoga with 11th gen cpu..