sebanc/brunch #746: Problems with the New Kernel (5.4) on SurfaceBook

JosephInn, 2020-12-26:
Hi! I just finished installing the latest framework release (r87), and after restarting it flashed the ChromeOS splash screen for a few seconds and then rebooted, I tried multiple times and the result is the same. Then I remembered that there is a new kernel (5.4), so went ahead and changed it back to the old kernel (4.19). It worked. 

I'm wondering if there is anything I could do to make the new kernel function with my hardware. Am I stuck with 4.19?

I also tried the new Unstable-Brunch-r87 release published yesterday with kernel 5.10, and got the same behavior (Splash-Screen and Rebooted) again.

Thanks for any Help! Great Project!

sebanc, 2020-12-26:
Hi, could you try the release in the brunch-unstable repo ? I fixed an issue which caused kernel panics on some devices.

sebanc, 2020-12-26:
Sorry just saw your latest comment saying you tried brunch-unstable, please dirsregard my previous comment.

sebanc, 2020-12-26:
I will look into it, it would be very useful if you could take a short video of the boot process in debug mode (without "console= vt.global_cursor_default=0 brunch_bootsplash=default" on the kernel command line)

JosephInn, 2020-12-26:
I hope this is what you meant:
[---> Images ](https://drive.google.com/file/d/1vD0IiuAMtoPastvjz_ExAtoJOpBeUwGt/view?usp=sharing)   (These are the key frames from the video, tried to make a bit more efficient for you)
[---> Video](https://drive.google.com/file/d/1UzJ09fFPZYmkP0C5BN_QfWs3sTFc0W9l/view?usp=sharing)

Also, Idk why but I didn't have "console= vt.global_cursor_default=0 brunch_bootsplash=default" on the kernel command line.

sebanc, 2020-12-26:
> I hope this is what you meant:
> [---> Images ](https://drive.google.com/file/d/1vD0IiuAMtoPastvjz_ExAtoJOpBeUwGt/view?usp=sharing) (These are the key frames from the video, tried to make a bit more [...]

It seems to be a kernel panic when loading modules, I have recently updated the surface SAM patch, could you try to boot with "module_blacklist=surface_aggregator" added on the kernel command line ?

> Also, Idk why but I didn't have "console= vt.global_cursor_default=0 brunch_bootsplash=default" on the kernel command line.

I updated the grub configuration to add a bootlogo, you can add those to your kernel configuration if you want the bootlogo.

JosephInn, 2020-12-26:
Tried it, but it didn't make any difference as far as I can tell.

> I updated the grub configuration to add a bootlogo

I like it, is very clean, I used "console=" before but this is definitely better.

Also while on this topic (Kernel Command line Parameters), I sometimes feel like I'm missing out, I've read all the ones on the Readme file, but I keep discovering others every time I talk to you, and so I was wondering if there any place where I could see more or maybe all of them (there's probably too many idk or probably they wouldn't be useful outside of some exceptional circumstance, but again idk)

sebanc, 2020-12-27:
I have been thinking about it and most hardware is common between my SP5 and your SB1 so I don't really see which module could generate a kernel panic on SB1. I am now considering it might be an issue with the data partition. Could you try a clean install on a USB flashdrive ?

Regarding kernel command line parameters, there are several types:
- most are mainline kernel command line parameters referenced here: https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html (e.g. console= or psmouse.synaptics_intertouch=X)
- a few are custom kernel command line parameters I have added (e.g. enforce_hyperthreading=X)
- others are not really kernel command lines parameters but just variables I pass to the initramfs through the kernel command line (e.g. options=X or img_part=X ...)

JosephInn, 2020-12-27:
I will try that, I'll get in touch when it's done.

There's something that I wasn't sure if I should of mentioned before, is a problem that happened immediately after I installed r87 and the "kernel panics" took place. ---> Whenever I go into another OS on my device I have to first disable "Secure Boot" or things usually dont work, I have to switch back and forth because I have BitLocker enabled (if I didn't switch it on then BitLocker would ask for ~30 digit key every time), so there I was with Secure Boot disabled and installing r87, after a couple tries and a couple crashes, I decided to go back into Windows for a bit, so I went into the UEFI and tried to enable Secure Boot, but it didn't work, and since then it hasn't worked, even after I went into windows (had to introduce the ~30 digit key and disabled BitLocker), and even after I changed the Kernel back to 4.19. I can live without BitLocker, but it was definitely something unexpected.

[This is the error ](https://drive.google.com/file/d/1V7FKZgjux79jrX9awJ2wcmpK30btojm8/view?usp=sharing)

JosephInn, 2020-12-27:
Did it, same issue, this time since it was a fresh install, it showed the setup screen for a second before rebooting.

sebanc, 2020-12-27:
Ok thanks for trying it. Could you try to boot with "usbcore.surface_tablet_mode=0" added to the kernel command line ?

JosephInn, 2020-12-27:
Still crashed, sorry

sebanc, 2020-12-27:
Could you try adding "module_blacklist=tpm_tis" on the kernel command line ?

JosephInn, 2020-12-27:
I was getting my hopes up while it rebuilt, but no, same result

sebanc, 2020-12-27:
I also had my hopes up, do you know why the system rebuilt itself ? could you post your grub config with "module_blacklist=tpm_tis" ?

JosephInn, 2020-12-27:
That second question made me take a good look at the grub file and re-test it. I left "module_blacklist=surface_aggregator" and "usbcore.surface_tablet_mode=0" thinking it wouldn't negatively affect the booting. But as it turns out just having "module_blacklist=tpm_tis" without the other two did actually solve the problem.

You did it! Thank you very much! you've been so patient and helpful.

Happy New Year in advance, you're a great dev.

closed 2020-12-27 by JosephInn

sebanc, 2020-12-28:
Happy to see that it worked :)