sebanc/brunch #747: MacBook Pro 2015 13-inch bug with android app sound

adelamint, 2020-12-27:
I used version 87 with Rammus image. Almost everything works including backlit keyboard but sound is not working once I launch and do something on Android application. For example, If I launch Google Play Store in middle of listening to music, the music will continue to play but without sound as if it's muted. Is it to do something with Sound Card driver or it is just a bug with Rammus image? I'm planning on trying version 86 with Eve image, though.

adelamint, 2020-12-27:
__Update: Tried version 86 from Eve. The issue persists. I think its driver built into Brunch is causing this issue.

sebanc, 2020-12-27:
Could you update to the latest release in the brunch-unstable repo ? Not sure if it is related but I fixed a sound issue that was present in the latest stable.

adelamint, 2020-12-27:
I tried unstable repo one, but sound issue persists. To recreate this issue, boot to live usb at least twice (Which will only show "Brunch Framework is loading" instead of creating / applying rootfs), then sound breaks whenever I try to run any android application.

adelamint, 2020-12-28:
> Could you update to the latest release in the brunch-unstable repo ? Not sure if it is related but I fixed a sound issue that was present in the latest stable.

Newest version of Brunch on MacBook Pro 2015 13-inch. The issue persists but I caught more detail. 

I guess OS "thinks" headphone is plugged in when android app is launched (and do at least one click within android app). Before running android application, MacBook's headphone jack glows in red, when at least one click is done within android app, the light on headphone jack shuts off and audio comes back after putting headphone into headphone jack then removing it again. (Weirdly, headphone jack does not glow right after pulling out headphones but once I play any sound it starts to glow again.

sebanc, 2020-12-28:
I am wondering if it could be related to the SPDIF issue in the below thread:
https://bbs.archlinux.org/viewtopic.php?pid=1492432#p1492432

Could you try to mute IEC958 as suggested in the edit part of the comment 11 ?

sebanc, 2020-12-28:
Otherwise, there is also a specific snd-hda-intel model for the macbook pro 2015. Could you try to add "snd_hda_intel.model=mbp81" on the kernel command line (in the grub configuration, after cros_debug)

adelamint, 2020-12-28:
> I am wondering if it could be related to the SPDIF issue in the below thread:
> https://bbs.archlinux.org/viewtopic.php?pid=1492432#p1492432
> 
> Could you try to mute IEC958 as suggested in the edit part of the comment 11 ?

It does not show up IEC958. If I try to execute that script, it will show "Wrong scontrol identifier: IEC958".
If I enter "amixer scontrols" then it returns only "Simple mixer control 'Master', 0.

Weirdly, at first boot, everything works fine. It starts to have issue after the second boot.

adelamint, 2020-12-28:
> Otherwise, there is also a specific snd-hda-intel model for the macbook pro 2015. Could you try to add "snd_hda_intel.model=mbp81" on the kernel command line (in the grub configuration, after cros_debug) ?

I also tried this, but makes no difference.

sebanc, 2020-12-28:
Could you attach the output of "lsmod" in crosh shell to this issue ?

adelamint, 2020-12-28:
If I just execute lsmod:
Module                  Size  Used by
vhost_vsock            20480  0
vmw_vsock_virtio_transport_common    24576  1 vhost_vsock
vsock                  36864  12 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  32768  1 vhost_vsock
applesmc               20480  0
input_polldev          16384  1 applesmc
acpi_als               16384  0
sbs                    16384  0
kfifo_buf              16384  1 acpi_als
sbshc                  16384  1 sbs
industrialio           65536  2 acpi_als,kfifo_buf
apple_bl               16384  0
snd_hda_codec_hdmi     53248  1
btusb                  61440  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
hid_apple              16384  0
bcm5974                20480  0
btintel                24576  1 btusb
brcmfmac              331776  0
brcmutil               16384  1 brcmfmac
snd_hda_codec_cirrus    20480  1
snd_hda_codec_generic    69632  1 snd_hda_codec_cirrus
thunderbolt           196608  0
ledtrig_audio          16384  1 snd_hda_codec_generic
mei_hdcp               16384  0
cfg80211              630784  1 brcmfmac
snd_hda_intel          36864  6
snd_intel_dspcfg       24576  1 snd_hda_intel
intel_pch_thermal      16384  0
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_cirrus
bdc_pci                16384  0
snd_hda_core           69632  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_cirrus

snd_hwdep              16384  1 snd_hda_codec
mei_me                 24576  1
mei                    69632  3 mei_hdcp,mei_me
i915                 1785856  42
video                  40960  1 i915
intel_gtt              20480  1 i915

If I run lsmod | grep snd
snd_hda_codec_hdmi     53248  1
snd_hda_codec_cirrus    20480  1
snd_hda_codec_generic    69632  1 snd_hda_codec_cirrus
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_intel          36864  6
snd_intel_dspcfg       24576  1 snd_hda_intel
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_cirrus
snd_hda_core           69632  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_cirrus
snd_hwdep              16384  1 snd_hda_codec

sebanc, 2020-12-28:
Could you try to add "modprobe snd_hda_codec_cirrus" in /etc/init/preui-probe.conf (under "modprobe i915") ?

sebanc, 2020-12-28:
Also, could you try kernel 4.19 and 5.10 ? (replace /kernel in the grub config respectively by /kernel-4.19 or /kernel-5.10)

adelamint, 2020-12-28:
Neither of options make sound on android apps work.

sebanc, 2020-12-29:
Sorry but I am out of ideas for now, maybe try to switch sound related chrome flags.

sebanc, 2020-12-30:
I thought of something, maybe you can try "options=internal_mic_fix" (even though it is not its main purpose, it fixes the sound card  order which might be related to your issue)

Pinoguino, 2021-01-14:
I have the same problem with the same macbook pro 2015 13 inch. I tried the options=internal_mic_fix but it doesn't fix the problem

sebanc, 2021-02-21:
I am not sure if macbook are supported by the new SOF kernel sound driver but it seems to be functional in latest brunch-unstable releases.

If someone following this issue is interested to try it, you can try using the latest "brunch-unstable" release and adding "snd_hda_intel.dmic_detect=1 module_blacklist=snd_hda_intel,snd_soc_skl" to the kernel command line.

Pinoguino, 2021-02-21:
I'm interested how i can try the latest brunch unstable ?

sebanc, 2021-02-21:
The brunch-unstable releases are here:
https://github.com/sebanc/brunch-unstable/releases

Pinoguino, 2021-02-21:
Thanks but with this brunch i created a new chrome.img but i face a bootloop

sebanc, 2021-02-21:
Could you try another clean install ? I was actually going to release this build as stable, I would like to be sure there is an issue with it.

Pinoguino, 2021-02-21:
Unfortunately i am trying to do a clean install. But when i boot in normal, the loading brunch image appears and then i have a black screen then a bootloop. When i go in debug mode i can the loading process but i have a bootloop too

sebanc, 2021-02-21:
Could you attach a short video of the boot process in debug mode ?

Pinoguino, 2021-02-21:
[](https://photos.app.goo.gl/8gzNsFQu5HNLvcRs5)

sebanc, 2021-02-21:
Unfortunately, I cannot really see the log lines on the video. Did you see the brunch rebuilding rootfs boot images on first boot ?

Pinoguino, 2021-02-21:
Yes

sebanc, 2021-02-21:
If you could a video close to the screen it would help debugging this. Else you can also try 5.10 kernel (replace "/kernel" by "/kernel-5.10" in the grub config.

Pinoguino, 2021-02-21:
I think it was blurry because it was in 360p you can retry and play the video in 1080p (the option is present now)

Pinoguino, 2021-02-21:
https://photos.app.goo.gl/JC4ud86PCKZPbb957

sebanc, 2021-02-21:
Thanks for the new video. It confirms that something went wrong with the install, could you re-download the recovery image and brunch andr try again ?

Pinoguino, 2021-02-21:
It finally work (for the new image), i have add the kernel command lines and rebooted. After that i can't turn up the volume

Pinoguino, 2021-02-21:
But one things i noticed is that the red light that came from the jack plug had totally disappear (kernel line added or not)

sebanc, 2021-02-22:
If I understand correctly, sound does not work with "snd_hda_intel.dmic_detect=1 module_blacklist=snd_hda_intel,snd_soc_skl" added on the kernel command line, right ? Is the android apps sound issue still present without those kernel command line parameters ?

Pinoguino, 2021-02-22:
Yes the sound does not work with the kernel line. And the the android apps sound issue is still present without those kernel line

Pinoguino, 2021-03-10:
I have the latest brunch unstable and today r89 was released. I updated my laptop and after the reboot the sound is working again in android app. I reboot my laptop multiple times to see but i don't know why now it's working

closed 2026-03-25 by github-actions[bot]