sebanc/brunch #748: HP Envy 2020

T-SLA, 2020-12-27:
Firstly, thank you for your amazing work.  You're doing an fantastic job.

I have a HP Envy 2020 ep0008na and have managed to get a triple boot system with Windows, Linux and Chrome OS.
I have updated to the latest Chrome OS (Version 87.0.4280.109) and Brunch (r87) and along with a few kernel options have got quite a few things working:

- wifi works after adding the intel option
- keyboard and trackpad work well after the last update
- power key works with the acpi_power_button option
- camera works great

However, a few essential things don't seem to work:
- the screen stays on max brightness.  The brightness hotkeys make the slider go up and down but the brightness doesn't change.  This laptop has a mega bright OLED screen so my retinas are burning!
- The audio - the hotkeys make the volume slider go up and down but there is no volume.  
- The touchscreen - does not work.  I tried the force tablet mode but no joy.

Could you help on any of these issues please?

sebanc, 2020-12-28:
Could you update to the latest release and confirm if you still have the same issues ? (there is a fix for audio on some devices and for some touchscreens)

Regarding brightness on OLED screens, could you try the 5.10 kernel in the latest release ? (replace /kernel in the grub configuration by /kernel 5.10)

T-SLA, 2020-12-28:
> Could you update to the latest release and confirm if you still have the same issues ? (there is a fix for audio on some devices and for some touchscreens)
>
> Regarding brightness on OLED screens, [...]

Hi, I updated to the latest build which has sorted the audio!  Thank you.  The touchscreen still does not work - I believe it is an Elan.  I  tried both default and 5.10 kernels and the brightness does not change.

sebanc, 2020-12-28:
Regarding the touchscreen, it is probably a mainline kernel issue, could you try a linux live usb and see if it works there ?

Regarding the OLED brightness, it is not supported by the mainline kernel, however there is a recent experimental patch series which aims to enable it but I need it to be ported to kernel 5.10:
https://aur.archlinux.org/packages/linux-oled

T-SLA, 2020-12-28:
The touchscreen and oled brightness work fine in Ubuntu if that helps?

sebanc, 2020-12-28:
Could you post a "dmesg" from ubuntu and one from chromeos ?

T-SLA, 2020-12-28:
Yes, no problem.  Attached.
[dmesgchrome.log](https://github.com/sebanc/brunch/files/5748747/dmesgchrome.log)
[dmesgubuntu.log](https://github.com/sebanc/brunch/files/5748748/dmesgubuntu.log)

sebanc, 2020-12-30:
I am working on OLED brightness, however the patches I found don't apply correctly on chromiumos kernel so it will be difficult.

Regarding the touchscreen it seems to be correctly recognized by the kernel, could you run "sudo evtest" in crosh shell, select the touchscreen, see if events are generated and post the output here ?

T-SLA, 2020-12-30:
> I am working on OLED brightness, however the patches I found don't apply correctly on chromiumos kernel so it will be difficult.
>
> Regarding the touchscreen it seems to be correctly recognized by the [...]

Sudo evtest gives the following options:
chronos@localhost / $ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      AT Translated Set 2 keyboard
/dev/input/event3:      HP Wide Vision HD Camera: HP Wi
/dev/input/event4:      HP WMI hotkeys
/dev/input/event5:      Video Bus
/dev/input/event6:      Video Bus
/dev/input/event7:      ELAN2513:00 04F3:2B75
/dev/input/event8:      ELAN2513:00 04F3:2B75 UNKNOWN
/dev/input/event9:      ELAN2513:00 04F3:2B75 UNKNOWN
/dev/input/event10:     SYNA32A6:00 06CB:CE24 Mouse
/dev/input/event11:     SYNA32A6:00 06CB:CE24 Touchpad
/dev/input/event12:     HDA NVidia HDMI/DP,pcm=3
/dev/input/event13:     HDA NVidia HDMI/DP,pcm=7
/dev/input/event14:     HDA NVidia HDMI/DP,pcm=8
/dev/input/event15:     HDA Intel PCH Mic
/dev/input/event16:     HDA Intel PCH Headphone
/dev/input/event17:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event18:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-21]: 

If I choose 7, I get some output when I touch the screen.  If I choose 8 or 9, I get nothing.  An example of the output from 7 is shown below:

Event: time 1609366418.479887, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 146100
Event: time 1609366418.479887, -------------- SYN_REPORT ------------
Event: time 1609366418.488181, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 154400
Event: time 1609366418.488181, -------------- SYN_REPORT ------------
Event: time 1609366418.496438, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 162500
Event: time 1609366418.496438, -------------- SYN_REPORT ------------
Event: time 1609366418.504689, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 170700
Event: time 1609366418.504689, -------------- SYN_REPORT ------------
Event: time 1609366418.513087, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 178900
Event: time 1609366418.513087, -------------- SYN_REPORT ------------
Event: time 1609366418.521291, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 187100
Event: time 1609366418.521291, -------------- SYN_REPORT ------------
Event: time 1609366418.529505, type 3 (EV_ABS), code 52 (ABS_MT_ORIENTATION), value 1
Event: time 1609366418.529505, type 3 (EV_ABS), code 49 (ABS_MT_TOUCH_MINOR), value 2
Event: time 1609366418.529505, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 195300
Event: time 1609366418.529505, -------------- SYN_REPORT ------------
Event: time 1609366418.537779, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 203500
Event: time 1609366418.537779, -------------- SYN_REPORT ------------
Event: time 1609366418.546149, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 211700
Event: time 1609366418.546149, -------------- SYN_REPORT ------------
Event: time 1609366418.554365, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 219900
Event: time 1609366418.554365, -------------- SYN_REPORT ------------
Event: time 1609366418.562630, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 228100
Event: time 1609366418.562630, -------------- SYN_REPORT ------------
Event: time 1609366418.570854, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 236200
Event: time 1609366418.570854, -------------- SYN_REPORT ------------
Event: time 1609366418.579270, type 3 (EV_ABS), code 52 (ABS_MT_ORIENTATION), value 0
Event: time 1609366418.579270, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 0
Event: time 1609366418.579270, type 3 (EV_ABS), code 49 (ABS_MT_TOUCH_MINOR), value 0
Event: time 1609366418.579270, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 244500
Event: time 1609366418.579270, -------------- SYN_REPORT ------------
Event: time 1609366418.587319, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1609366418.587319, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1609366418.587319, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 252500
Event: time 1609366418.587319, -------------- SYN_REPORT ------------

T-SLA, 2020-12-30:
Is there a chance of the OLED brightness working?
Or the fingerprint sensor?

sebanc, 2020-12-31:
WRT OLED brightness, could you try the release in brunch-unstable repo ?

For touchscreen, could you run "sudo evdev", select the device "/dev/input/event7: ELAN2513:00 04F3:2B75", touch the 4 corners of the screen and post the full output ?

T-SLA, 2020-12-31:
OLED brightness works with the new unstable release.  However, it can be reduced all the way down to zero so the screen goes black.  But I'm happy that it's usable now, thank you. :-)

sebanc, 2020-12-31:
Nice for the brightness.

try "sudo evtest" for the touchscreen, it was a typo.

T-SLA, 2020-12-31:
Thank you so much for sorting the brightness.  Slowly but surely getting there and I really appreciate your time, effort and expertise.

I have attached the output you asked for.  I touched the top right corner, then bottom right, then bottom left, then top left.

[log.txt](https://github.com/sebanc/brunch/files/5757304/log.txt)

sebanc, 2020-12-31:
Unfortunately the evtest log looks perfect, I have no idea why your touchscreen does not work...

T-SLA, 2021-01-01:
OK, so I got the touchscreen to work by the following method - I hope it can help someone:

1- goto chrome://flags and search for ash-debug-shortcuts and enable it.  Restart
2 - press windows+shift+t to enable the touchscreen - the same key combo will toggle the touchscreen on and off

sebanc, 2021-01-01:
@T-SLA ash-debug-shortcuts is enabled by default in brunch, you had probably pressed windows+shift+t by mistake previously and pressing it again re-enabled the touchscreen.

T-SLA, 2021-01-01:
Is there something that will help to make the battery last longer?  I have dual GPUs and have used the kernel line "module_blacklist=nouveau" to disable the nvidia.  However, it runs hot, fan on all the  time and I get about 1.5 hours battery.

sebanc, 2021-01-01:
acpi_call module is included in brunch, you can enable it by running:
`sudo insmod /lib/modules/$(cat /proc/version |  cut -d' ' -f3)/acpi_call.ko`

Then use this script which contains some common dGPU power off methods to detect the one your device use:
https://github.com/mkottman/acpi_call/blob/master/examples/turn_off_gpu.shhttps://github.com/mkottman/acpi_call

Once you found the right one, run `echo '< ACPI method to disable the dgpu e.g. _SB.PCI0.PEG1.GFX0.DOFF >' > /proc/acpi/call` and see if it works.

If it does work, I post the method here and I will explain how to automate it on boot.

Alternatively, you can install the brunch-toolchain and build the "bbswitch" module.

T-SLA, 2021-01-01:
Thank you.  The link for the script is "Page Not Found".  Is there another link please?

sebanc, 2021-01-01:
indeed the link was incorrect, I edited it in my above post

T-SLA, 2021-01-02:
Thank you.  I keep getting the below error when I run the script.  I have done chmod +x but still get the same error...

bash: ./turn_off_gpu.sh: Permission denied

Can you help?

sebanc, 2021-01-02:
Not sure why it does not work but you can copy turn_off_gpu.sh to /usr/bin/.
And then run "sudo /usr/bin/turn_off_gpu.sh"

T-SLA, 2021-01-02:
Great, that worked.  The only one that showed as "works!" was \_SB.PCI0.PEG0.PEGP._OFF

And then the system froze....

sebanc, 2021-01-02:
Run the 3 below commands in crosh shell to disable the gpu on boot:
- `sudo mkdir -p /var/brunch/bootscripts`
- `echo "echo '_SB.PCI0.PEG0.PEGP._OFF' > /proc/acpi/call' | sudo tee /var/brunch/bootscripts/disable_dgpu.sh`
- `sudo chmod 0755 /var/brunch/bootscripts/disable_dgpu.sh`

T-SLA, 2021-01-02:
Thank you.  It wasn't working but I think there may be a typo - should it be " instead of ' after the word call?

sebanc, 2021-01-02:
Indeed it should be a ", I edited it

sebanc, 2021-02-07:
Fyi, the OLED brightness patch introduced a regression for some users, starting from the next release you will have to enable it using "options=oled_display"