sebanc/brunch #751: Change kernel (from 5.4 to 4.19): Login to Chrome OS not possible

saturn-hh, 2020-12-28:
expected behaviour: Change of kernel version without problems and usable system
actual behaviour:      After automatic rebuild login was not possible. Only a reset of password AND user data
                                enabled access to the system. Error message: "... wrong password"
Reproduce the error: Install the system and after first login and connecting to the google account changing
                                  the kernel version on the grub command line.

First I would like to say thank you for the great work and the opportunity to use my surface 2017 with chrome os. In the first place everything went smooth and I could install chrome os on a separate partition (ext 4) on my surface pro 2017 (Version 5). The system was up and running. 

Due to the known limitations of the new touchscreen driver (ipts user space daemon) I wanted to try the 4.19 kernel and changed the kernel command line accordingly. After the reboot the brunch framework rebuild parts of the system and after that I saw the usual login screen. Unfortunately I couldn't login and got the error message "Your password couldn't ..." -> wrong password. Even disabling the two factor authentification didn't  help. I sill could login as guest and could login to my google account in the browser, but I couldn't login to the chrome os with the actual user. 

The only way to get access to the system was to reset my google password in combination with an power wash of the installed system. So the user data where gone.

Any advice would be appreciated.

Alexander

Chrome OS Version: 87 (chromeos_13505.73.0_rammus_recovery_stable-channel_mp-v2)
brunch Framework Version: Brunch r87 stable 20201216
Hardware: Microsoft Surface 5 (2017) 
Dual boot with Linux Mint 20.04

MohamedAlmahmood, 2020-12-28:
I also have the same problem. I downgraded to kernel 4.19 as the touchscreen on the newest kernel is very unstable. Now I'm signing in using my other google account. This is very annoying as I am no longer the admin. 
Hardware: Microsoft Surface 5 (2017)
Dual boot with Windows

bubibubi, 2020-12-30:
I think the issue is not related to surface. I'm using an ASUS Transformer and after switching to 4.19 I can't log in anymore.
Investigating, it seems that the network is broken with 4.19.

sebanc, 2020-12-30:
Just a comment to say that this issue is not systematic, I swap kernel regularly after login in with my google account and have not faced this so far.

I have no idea what might be the reason for this:
- a specific extension/app installed in chromeos/crostini
- an issue with the chrome sync profile
- something else

If you want to try resetting chrome sync:
Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

AmroLouwe, 2020-12-30:
I am having issues with touchscreen on my surface book too. Been trying to get chromeos running on it with touchscreen, hasn't been successful yet

sebanc, 2021-01-01:
It has been reported that after 2 failed login attempts you should be able to login and resetting chrome flags should then fix this.

saturn-hh, 2021-01-27:
UPDATE: After reset the chrome os on the surface I could login again. 

Today I changed an Boot Option and *suprise* the "rebuild" worked. I habe no idea what was different Form my first try.

closed 2021-01-27 by saturn-hh