sebanc/brunch #753: Does chrome os will be stable on my pc. And which should be best recovery for me?

yegender124001, 2020-12-28:
Hi, @sebanc I only just want to ask that does chrome os will be at stable condition on my device. I have tried Chrome os dual boot. And think to install it on hard disk.

My PC Specs:

Processor:i3 2nd gen
Motherboard: DH61HO
Hard Disk : 1 TB
Wireless Dell Mouse and Keyboard
Monitor: DELLIN1910

I have tried never ware cloudready on it. It works well on it. I have tried Chrome os also. It also worked fine just with some lags and android issue. I think that it may be not after installation to hard disk. I install it because I have only this option. Please tell me that does it will be stable. 

Thanks

Your Friend

sar777-dev, 2020-12-28:
Why not try asking  "brunch-toolkit" from WesBosch? that has compatibility check feature which detects the recovery image to use https://github.com/WesBosch/brunch-toolkit

yegender124001, 2020-12-28:
Thanks. I only just want to use Chrome os for my studies. I think it will work best for me. I need stability to just to get no intrupption in my study.

yegender124001, 2020-12-28:
Thanks. I only just want to use Chrome os for my studies. I think it will work best for me. I need stability to just no intrupption in my studies

yegender124001, 2020-12-28:
Does this script will work on never ware cloudready?

sar777-dev, 2020-12-28:
I didn't tried it on neverware cloudready but you can it a try following through instructions provided on Wesbosch readme page, or why don't live boot any linux distro and install 'brunch-toolkit' and check you compatibility with your hardware.

yegender124001, 2020-12-29:
> I didn't tried it on neverware cloudready but you can it a try following through instructions provided on 'Wesbosch' readme page, or why don't live boot any linux distro and install 'brunch-toolkit' and check you compatibility with your hardware.

Does this will install chrome os in hard disk without a USB without making a img file ?

sar777-dev, 2020-12-30:
Yes, you can also use 'brunch-toolkit' to install brunch ChromeOS on your hard disk/SSD.

yegender124001, 2020-12-30:
> Yes, you can also use 'brunch-toolkit' to install brunch ChromeOS on your hard disk/SSD.

Thanks. I will try it.

yegender124001, 2020-12-30:
Hi, I have installed it using brunch-toolkit. It works. I always tries Samus image for my PC. But it tells that compitable is another. Thanks for it.

yegender124001, 2020-12-30:
But now It has some lag. And play store is slow in working.

closed 2020-12-31 by yegender124001