sebanc/brunch #754: Youtube "Glitch" lines

VivihugoFranz, 2020-12-28:
Hey, first of all, sorry for bad English, it's not my native language.
Hello to everyone.
I recently started to use Chrome OS in my computer with Brunch, and it works wonderfull. Thanks a lot for this project.
But, almost every time I'm on Youtube, some lines apears on the bottom of the video. I have no idea why it's happening, and I did'nt see anyone with the same problem. 
[image]
My laptop configs are:
Dell Latitude E5440
i5 4310U
8GB RAM
Running ChromOS 87.0.4280.109 
Brunch r87 stable 20201227

sebanc, 2020-12-28:
It seems to be a known chrome bug with your gpu, try changing the zoom percentage in chrome or the display size in settings.

VivihugoFranz, 2020-12-28:
Yeah, I checked it. I'm new into "Non Windows world", so i tought that was chrome OS problem. I look for that, thanks.

closed 2020-12-28 by VivihugoFranz