sebanc/brunch #755: Save ALSAMIXER configs

VivihugoFranz, 2020-12-28:
Hey, first of all, sorry for bad English, it's not my native language.
Hello to everyone.
I recently started to use Chrome OS in my computer with Brunch, and it works wonderfull. Thanks a lot for this project.
But, everytime i start the system, my microphone dosn't work. So, I have to open the Alsamixer configuration in shell, to set it.
I have tried the command  `sudo alsactl store` after set the mic on, it save the configs, but when I restart the system, it back to the default.
It is curius because, the system really saved the configs. When I type `sudo alsactl restore`, the system restore the exactly same config that I made previsiusly. 
But it dosn't seem to load when I start the system.
[image]

Again, sorry for any erros in writing.
Can anyone help me?

My laptop configs are:
Dell Latitude E5440
i5 4310U
8GB RAM
Running ChromOS 87.0.4280.109
Brunch r87 stable 20201227

sebanc, 2020-12-28:
Try to run the following commands:
```
sudo mkdir -p /var/brunch/bootscripts
echo "alsactl restore" | sudo tee /var/brunch/bootscripts/alsa.sh
sudo chmod 0755 /var/brunch/bootscripts/alsa.sh
```

VivihugoFranz, 2020-12-28:
> Try to run the following commands:
> 
> ```
> sudo mkdir -p /var/brunch/bootscripts
> echo "alsactl restore" | sudo tee /var/brunch/bootscripts/alsa.sh
> sudo chmod 0755 /var/brunch/bootscripts/alsa.sh
> ```
> 
> Then reboot and check the settings

Ir worked! Thanks a lot, I spent a lot of time tring to make this work.

closed 2020-12-28 by VivihugoFranz