sebanc/brunch #756: Stuck at loading brunch framework

AryanPandeyy, 2020-12-29:
[image]

I tried to install Chrome OS dual booting with Linux Mint. 
I am trying to boot Chrome os image using grub2
But the moment it start to boot
It shows "loading brunch framework "
And stuck at this message forever.

sebanc, 2020-12-30:
Could you boot without "console= vt.global_cursor_default=0 brunch_bootsplash=default" in the grub configuration and attach a short video of the boot process ?

Also, what's your laptop model (or cpu/gpu if it is a desktop) ?

AryanPandeyy, 2021-01-02:
I have AMD E1-1200 APU with Radeon(tm) HD Graphics

I had also attached PC.txt file which contain all information
regarding information regarding my PC

Thanks you for helping me.

AMD E1-1200 APU with Radeon(tm) HD Graphics

-------------------------------------------------------------------------

SMBIOS Version			2.7

DMI BIOS		
	vendor			AMI
	version			8.12
	date			03/21/2013
	ROM size		1024 KB

DMI System Information		
	manufacturer		Hewlett-Packard
	product			18-1106in
	version			unknown
	serial			5CM32500X0
	UUID			{37853A4D-5E3B-A684-DF2F-81F26EC4DDDB}
	SKU			H4J18AA#ACJ
	family			103C_53316J G=D

DMI Baseboard		
	vendor			Hewlett-Packard
	model			2AF2
	revision		A01
	serial			5CM32500X0

DMI System Enclosure		
	manufacturer		Hewlett-Packard
	chassis type		Desktop
	chassis serial		5CM32500X0

DMI Extension Slot		
	designation		J6B2
	type			A5
	populated		yes

DMI Extension Slot		
	designation		J6B1
	type			A5
	populated		yes

DMI Extension Slot		
	designation		J6D1
	type			A5
	populated		yes

DMI OEM Strings		
	string[0]		FBYTE#2V333E3K3N3Q4h5c6J6S6b7B7K7P7Q7S7W7a7s9S9ha3apaqarau.8Y;BU
	string[1]		ILDID#13AP1AR8603#SACJ#DACJ;
	string[2]		 
	string[3]		 
	string[4]		 
	string[5]		 
	string[6]		 
	string[7]		 
	string[8]		 
	string[9]		 
	string[10]		 
	string[11]		 
	string[12]		 
	string[13]		 
	string[14]		 
	string[15]		 

DMI Physical Memory Array		
	location		Motherboard
	usage			System Memory
	correction		None
	max capacity		2 GB
	max# of devices		2

DMI Memory Device		
	designation		A1_DIMM0
	format			DIMM
	type			unknown
	data width		64 bits

DMI Memory Device		
	designation		A1_DIMM1
	format			DIMM
	type			DDR3
	total width		64 bits
	data width		64 bits
	size			2 GB

DMI Processor		
	manufacturer		AuthenticAMD
	model			AMD E1-1200 APU with Radeon(tm) HD Graphics
	clock speed		1400.0 MHz
	FSB speed		100.0 MHz
	multiplier		14.0x

Storage
-------------------------------------------------------------------------

Drive	0
	Name			WDC WD5000AAKX-60U6AA0
	Revision		18.01H18
	Serial			     WD-WCC2EEV65531
	Capacity		465.8 GB
	Type			Fixed
	Bus Type		SATA (11)
	Rotation speed		7200 RPM
	Features		SMART
	Volume			c:\, 465.3 GBytes (95.9 percent available)

USB Devices
-------------------------------------------------------------------------

USB Device			Generic USB Hub, class=0x09, subclass=0x00, vendor=0x05E3, product=0x0606
USB Device			USB Composite Device, class=0x00, subclass=0x00, vendor=0x2A7A, product=0x9A18
USB Device			USB Input Device, class=0x00, subclass=0x00, vendor=0x275D, product=0x0BA6
USB Device			USB Composite Device, class=0xEF, subclass=0x02, vendor=0x04F2, product=0xB34A

Graphics
-------------------------------------------------------------------------

Number of adapters		1

Graphic APIs
-------------------------------------------------------------------------

API				D3D
API				ATI I/O
API				ADL SDK

Display Adapters
-------------------------------------------------------------------------

Display adapter 0	
	ID			0x2000000
	Name			AMD Radeon HD 7310 Graphics
	Board Manufacturer	Hewlett-Packard
	Cores			0
	ROP Units		0
	Memory size		384 MB
	PCI device		bus 0 (0x0), device 1 (0x1), function 0 (0x0)
	Vendor ID		0x1002 (0x103C)
	Model ID		0x9809 (0x2AF2)
	Revision ID		0x0
	Performance Level	0
		Core clock	280.0 MHz
		Memory clock	533.0 MHz

Win32_VideoController		AdapterRAM = 0x18000000 (402653184)
Win32_VideoController		DriverVersion = 15.200.1062.1004
Win32_VideoController		DriverDate = 08/03/2015

Monitor 0	
	Model			HP All in One (Hewlett Packard)
	ID			HWP410E
	Serial			
	Manufacturing Date	Week 1, Year 2011
	Size			18.5 inches
	Max Resolution		1366 x 768 @ 60 Hz
	Horizontal Freq. Range	0-0 kHz
	Vertical Freq. Range	0-0 Hz
	Max Pixel Clock		0 MHz
	Gamma Factor		2.2

zakk87, 2021-01-09:
@AryanPandeyy Hi, I have the same issue on Lenovo G505 AMD E1-1200 APU. Did you resolve that problem?

fi5ti, 2021-01-19:
> Could you boot without "console= vt.global_cursor_default=0 brunch_bootsplash=default" in the grub configuration and attach a short video of the boot process ?
> 
> Also, what's your laptop model (or cpu/gpu if it is a desktop) ?

whenever i removed the console line it would say "GNU Grub is now loading Chrome OS", and then not do anything. for me, i have "rmmod tpm" at the top of the code. and removing that line will just get it stuck on loading brunch framework. but keeping it gets it stuck on the chrome logo.

RockyZone, 2021-01-29:
I am too stuck on the loading brunch framework,I use a YouTube tutorial to install it.This one"https://youtu.be/ROBpXNonVxc".
Any solution plz?

litszwaiboris, 2021-02-03:
You installed brunch framework at not the same recovery version
etc rammus recovery version 87 need to combine with version 87 of brunch

101201120, 2021-02-08:
凌动D2550处理器应该装哪个版本的安装框架和恢复镜像搭配使用可以成功呢

smartguy1196, 2021-06-06:
Ok. So I removed the mentioned code from the grub config.

This is what my screen is stuck on:

[image]

Hardware (a bit ambitious):

Motherboard - ROG Strix x570-E
CPU - Ryzen 7 5700
RAM - 64gb ddr4
Storage - 1TB nvme (samsung 980)
GPU - rx 6700xt

Running triple boot on the nvme

smartguy1196, 2021-06-06:
Hmm... I didn't change my kernel version. Let me try that

Napoleon99999, 2021-10-02:
How to change kernel please 🥺

AryanPandeyy, 2022-01-13:
> @AryanPandeyy Hi, I have the same issue on Lenovo G505 AMD E1-1200 APU. Did you resolve that problem? 

No