sebanc/brunch #759: ASUS Transformer T101HA Mic and Cam not working (brunch r87 w/ kernel 4.19)

bubibubi, 2020-12-30:
I'm using kernel 4.19 so I solved an issue about brightness regulation but Mic and Cam (and BT) still not working.
Is there something I could try?

jakino0, 2021-01-01:
also having t101ha i have the same problems. I am available for tests

sebanc, 2021-01-02:
Sorry but both camera and microphone on this model are not supported by the linux mainline kernel.

bubibubi, 2021-01-03:
I'm also having a problem with the external ear phones (kernel 4.19).
Checking everything with linux mint 20, external earphones should work.

Brigthness regulation working only in 4.19
Mic not working (nor in mint)
Cam not working (nor in mint)
Rotation (sensors?) is not working (but is not working also using android apps while it's working using Chrome OS feature)
Audio working
(during boot the screen is rotated, same in mint)
 
Any ideas about ear phones?

jakino0, 2021-01-04:
> Any ideas about ear phones?
 `sudo nano /usr/share/alsa/ucm/chtrt5645/HiFi.conf`.
Edit the lines cset "name='Headphone Switch' off" and cset "name='Headphone Channel Switch' off" to 'on'.

bubibubi, 2022-05-09:
Update for r100

Mic & Cam not working (but I'm still using the kernel 4.19)

Rotation sensors work but there is a mismatch of 90 degree from the position of the tablet and the display orientation. Probably there is a configuration somewhere but I can't find it.
The trick for the headphones does not work anymore

boccafos, 2022-06-20:
Hi, i don't know very much of linux and similar but i'm writing because i tried to install chrome os through brunch in my asus t100ha. I have rammus version 97 and almost everything works apart of camera and microphone that are really important for me. Any suggestions?