sebanc/brunch #761: Blank screen after chrome os logo

yegender124001, 2020-12-31:
Hi @sebanc I have installed chrome os on vmware. I got boot splach. Then a blank screen. Please see this

[image]
[image]

Then after
[image]

I think it can be installed on a vmware virtualmachine by your future updates of brunch

Thanks

ameyjoshi3068, 2020-12-31:
Chrome OS cannot be installed on the virtual machine due to a lack of graphics drivers as mentioned [here](https://github.com/sebanc/brunch/issues/303#issue-637525469) by sebanc

If you are interested you can try adding them youreslf.

yegender124001, 2021-01-03:
OK I will try to add them.

closed 2021-01-03 by yegender124001

Jhone-Harry, 2022-11-13:
Hello, Installing ChromeOS on VMware Workstation Pro is what I want to do. There is a statement that "visual machines are not supported" on the GitHub downloads page. Is it possible to make work on VMware Workstation or VirtualBox?
Please, if you can assist, I appreciate it.

Jhone-Harry, 2022-11-13:
Hello, @yegender124001 Installing ChromeOS on VMware Workstation Pro is what I want to do. **a success for you? Have you added the drivers?**
Please, if you can assist, I appreciate it.

yegender124001, 2022-11-18:
> Hello, @yegender124001 Installing ChromeOS on VMware Workstation Pro is what I want to do. **a success for you? Have you added the drivers?** Please, if you can assist, I appreciate it.

Cloudready aka Chrome OS Flex Works but not chrome OS when i tried. Chromium OS works too. But i didn't tried installing drivers