sebanc/brunch #764: [HELP and TEST needed] Tablet mode detection with Intel VBTN

midi1996, 2020-12-31:
Currently the way brunch detects if it should enable tablet mode on Chrome OS is by checking either the USB IDs found in `99-tablet_mode.sh` or use your own as stated in `99-hid_tablet_mode.sh`.
Intel Virtual Button is a device that Windows uses for many actions like "Modern Sleep" power button, volume keys and tablet mode detection among other things. Some other devices (like Lenovo Yoga and some Thinkpads with switchable modes) have another device that outputs those modes through `thinkpad_acpi` events.

This thread is about creating a way to use this detection method when possible (as it's probably more general and standardized).

I started checking this repo https://github.com/alesguzik/linux_detect_tablet_mode/ which is mainly for thinkpads but also had information about other devices (like the Dell XPS 9365) which use Intel VBTN.

My first test is with this script:

```bash
#!/bin/bash

device='/dev/input/by-path/pci-0000:00:1f.0-platform-INT33D6:00-event'
event_tablet_on='value cc'
event_tablet_off='value cd'

evtest "$device" | while read line; do
	case $line in
		*"$event_tablet_off"* ) echo "TABET_OFF"; echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode ;;
		*"$event_tablet_on"* ) echo "TABLET_ON"; echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode ;;
	esac
done
```

I used values that I got from my device (HP Elite X2 1012 G2). I would like you to test and share your values with me to see if there is any pattern we can use to create a better solution.

**Issues I'm facing:**

- [ ]  No way to detect the current state of the device<sup>1</sup>

1: At boot, I have no way to detect if the device is in tablet mode or not until I physically move the keyboard. As far as my research goes I've seen some posts requiring the usage of `libinput` which isn't implemented in Chrome OS to begin with.

---

**How to test:** (for people with intel_vbtn, no idea for thinkpad users)

1. Boot into a fairly recent linux distribution
2. Check if `intel_vbtn` is loaded in `lsmod`
    * You can also check  `evtest` by running it and you'll get a list with `Intel Virtual Button driver` entry (remember its number)
    * You can also check by running `cat /proc/bus/input/devices | grep -i "virtual button"`
3. Run `evtest` and then type the number of the `Intel Virtual Button driver` event
4. Detach/attach or flip your keyboard and get the log, remember which values are printed and when
5. Share the information here.

**Information to share:**

* Laptop model: [OEM] [Full model]
* Has intel_vbtn: Yes/No
* Values when in laptop mode:
```
Your results here
```
* Values when in tablet/convertible mode:
```
Your results here
```

Example:

* Laptop mode: HP Elite X2 1012 G2
* Has intel_vbtn: Yes
* Values when in laptop mode:
```
Event: time 1609431377.708078, type 4 (EV_MSC), code 4 (MSC_SCAN), value cd
Event: time 1609431377.708078, -------------- SYN_REPORT ------------
```
* Values when in tablet mode:
```
Event: time 1609431352.997069, type 4 (EV_MSC), code 4 (MSC_SCAN), value cc
Event: time 1609431352.997069, type 1 (EV_KEY), code 240 (KEY_UNKNOWN), value 1
Event: time 1609431352.997069, -------------- SYN_REPORT ------------
Event: time 1609431352.997123, type 1 (EV_KEY), code 240 (KEY_UNKNOWN), value 0
Event: time 1609431352.997123, -------------- SYN_REPORT ------------
```
---

Any help and suggestions is welcome.

sebanc, 2021-01-01:
Thanks to your investigations, I think I can fix this properly.

Could you post the output of "cat /sys/class/dmi/id/chassis_type" and attach a dmesg ?

midi1996, 2021-01-01:
[dmesg_log.txt](https://github.com/sebanc/brunch/files/5759096/dmesg_log.txt)

here si the dmesg

and the chassis type is `30`

sebanc, 2021-01-01:
Could you try tablet mode on the new brunch-unstable build (without any tablet mode framework option) ?

librehat, 2022-06-20:
@sebanc: I'm more than happy to test this with a slightly different device (Samsung Galaxy Book Pro 360).

`cat /sys/class/dmi/id/chassis_type` shows `31`. The latest r101 brunch framework doesn't switch on tablet mode when folded (however the keyword would be deactivated).

I also find that `lsmod` doesn't have `intel_vbtn` module loaded (Kernel 5.15)

librehat, 2022-06-20:
What I found out is that
`/dev/input/by-path/platform-INTC1051:00-event` will have an event with `code=1, value=1` when tablet mode should be activated, and `code=1, value=0` when it should be deactivated.

`/dev/input/by-path/platform-INTC1051:00-event` seems also used in Samsung Galaxy Book Flex (mine is Galaxy Book Pro 360. It might be a fair guess that it's also used for other convertible Samsung Galaxy Book products)

librehat, 2022-06-20:
I wrote a small C program that can trigger the tablet mode automatically on my Samsung Galaxy Book Pro 360 (hopefully this can provide a good clue to enable tablet mode from brunch framework).

```c
#include <stdlib.h>
#include <unistd.h>
#include <fcntl.h>
#include <errno.h>
#include <linux/input.h>
#include <string.h>
#include <stdio.h>

int main(void)
{
    const char *dev = "/dev/input/by-path/platform-INTC1051:00-event";
    const char *tablet_mode_dev = "/sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode";
    struct input_event ev;
    ssize_t n;
    int fd;

    fd = open(dev, O_RDONLY);
    if (fd == -1) {
        fprintf(stderr, "Cannot open %s: %s.\n", dev, strerror(errno));
        return EXIT_FAILURE;
    }
    FILE *t_fd = fopen(tablet_mode_dev, "w");
    if (t_fd == NULL) {
        fprintf(stderr, "Cannot open %s: %s.\n", tablet_mode_dev, strerror(errno));
        return EXIT_FAILURE;
    }
    
    while (1) {
        n = read(fd, &ev, sizeof ev);
        if (n == (ssize_t)-1) {
            if (errno == EINTR)
                continue;
            else
                break;
        } else
        if (n != sizeof ev) {
            errno = EIO;
            break;
        }
        printf("Value=%d 0x%04x (%d)\n", ev.value, (int)ev.code, (int)ev.code);
        if (ev.code == 1) {  // tablet mode switch
            if (fprintf(t_fd, "%d", ev.value) < 0) {
                fprintf(stderr, "Failed to update tablet mode\n");
            }
            fflush(t_fd);
            printf("Tablet mode switch %d\n", ev.value);
        }
    }
    fflush(stdout);
    fprintf(stderr, "%s.\n", strerror(errno));
    return EXIT_FAILURE;
}
```

For people trying to use it in the meantime, make sure you change the permission of `/sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode` so it doesn't require root privilege.