sebanc/brunch #766: "not a valid disk name" error nvme0n1p6

piotrus22, 2021-01-02:
hello - i am trying to install chrome os on a partition on my nvme ssd. following all the instructions, I keep getting an error "nvme0n1p6 is not a valid disk name". where nvme0n1p6 is the partition i'd like to install it on.

i was able to successfully install chrome os on a usb stick, and when I attempt to install the os from chrome, i get the same error message.

what am i missing here?

thank you in advance!

sebanc, 2021-01-02:
You cannot install chromeos as single boot to a partition, only to a full disk (nvme0n1). What you need to do to install chromeos to a partition is to follow the dual boot procedure which will create an image file within the nvme0n1p6 partition.

Try to follow the dual boot guide and let me know if your stuck at some specific step.

piotrus22, 2021-01-02:
thank you - the issue was i had to use the terminal to temporarily mount the nvme partition using the /tmpmount command. after that i was able to install the image and successfully booted. cheers mate!

closed 2021-01-02 by piotrus22