sebanc/brunch #767: Play store takes forever to open even after applying android_init_fix

BigFhungus, 2021-01-02:
My play store and all play store apps took forever to open (15-30 minutes), so I applied the android_init_fix, and still the problem persists, any suggestions?

sebanc, 2021-01-03:
Sorry but I have never been able to fix this issue which affects a few devices.

You could try to reset Chrome sync but I am not sure it will help in this case:
Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

yegender124001, 2021-01-09:
You should try the this https://github.com/WesBosch/brunch-toolkit
And then use that image which is in the output of this script. It may work.
I use samus and I have this issue but when I use this script it gives output to me that the recovery image is other.

BigFhungus, 2021-01-09:
Thank you for your response, I will try this out.

khanhtranngoccva, 2021-01-16:
Have you waited one minute on the login screen? What are your PC specs?

sebanc, 2021-01-16:
Honestly, I have no idea why this happens on some laptops. I am willing to try something else but I would need "lsmod" logs taken from the latest brunch release. (run "lsmod > ~/Downloads/lsmod.txt" in crosh shell and attach lsmod.txt from your downloads folder).

BigFhungus, 2021-01-17:
Thank you for the suggestions, but I have uninstalled brunch from my laptop

closed 2021-01-17 by BigFhungus