sebanc/brunch #769: No space left on device during install

busybox11, 2021-01-02:
Hey,
I'm trying to install brunch to a blank 32GB USB stick.
Unfortunately, the 3rd partition stops at 67% saying that there's no space left on this device (see screenshot below).
[image]

Here is the output of the created partition table:
[image]

And here is how my GParted looks like:
[image]
(Don't pay attention to the last unallocated space, it's because of my manipulation, I'm trying to grow the ROOT-C partition manually).

I'm using `brunch_r87_stable_20201227.tar.gz`, `chromeos_13505.73.0_grunt_recovery_stable-channel_mp-v6.bin.zip` on an AMD A9-9425.
Here is my `brunch` folder.
[image]

Thanks for this awesome project and I hope you'll be able to help me!

busybox11, 2021-01-02:
I tried to run it again and now for some reason it does not do the same thing.
It's still not working tho.
[image]

sebanc, 2021-01-02:
Reboot your computer and try again

busybox11, 2021-01-02:
Reboot? Yeah, I can do that. Why would that be a problem tho?

sebanc, 2021-01-02:
The partition table of the usb flashdrive could not be properly updated which prevents flashing partitions, a reboot should fix it

busybox11, 2021-01-02:
Okay, I'm going to do it ASAP then. I'll keep you updated on that. Thanks for your help!

busybox11, 2021-01-02:
[image]
Oh yeah, it seems that it's working now. I'll keep you updated.

busybox11, 2021-01-02:
[image]
Yeah, it worked. I apologize for this useless issue, thank you for your time anyway!
Thanks again for this incredible project :)

closed 2021-01-02 by busybox11

sebanc, 2021-01-02:
No pb :)