sebanc/brunch #770: Stuck on Chrome Logo - Desktop : MSI B360M Mortar, i5-9400F, RX570

kwmlman, 2021-01-03:
First of all, thank you for your wonderful work. ^^

I am trying chromeOS r87 on one hard drive. (My 9400F does not support Intel Graphics.)

**Terminal Work => Reboot => Stuck on Chrome Logo**

Can this problem be solved? If it doesn't work now, is it possible in the future?

sebanc, 2021-01-03:
Unfortunately no, chromeos recovery images of intel devices have no graphics driver for AMD gpu and that's not really something that I can change.

kwmlman, 2021-01-03:
Thank you for your warm reply.

I hope that you will be Bill Gates who upgraded the world. ^^

tejasraman, 2024-12-25:
> Unfortunately no, chromeos recovery images of intel devices have no graphics driver for AMD gpu and that's not really something that I can change.

is that the case on amd images as well? do AMD images now lack polaris (rx5xx) gpu drivers?

I have tried zork, skyrim, and guybrush images, and all of them boot to the chromeOS logo and dn not go any further. Verbose boot doesn't indicate an error.

However, on a PC with integrated Vega graphics (tested on 3400G and 2400GE) all 3 of these images boot to a setup screen and allow the user to continue setup.

The PC in question has a Ryzen 5 3600, RX570, 48GB RAM, and Brunch is installed on a tertiary 256GB NVMe SSD.