sebanc/brunch #771: 华硕X201E的机器，处理器是celeron847，google play一直都是something went wrong

oukaitou2020, 2021-01-03:
google play一直都是提示something went wrong,请大神帮忙解决啊，

sebanc, 2021-01-03:
I would suggest:
1) Try the latest brunch-unstable, I changed something which might have an impact on your device.
2) Reset chrome sync: Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

If it still does not work, it might be due to your VPN as I have seen it in other play store related issues

oukaitou2020, 2021-01-03:
Thanks, I'll try the latest version first,then I reply on this issue,hope to fix the problem.

oukaitou2020, 2021-01-05:
> I would suggest:
> 1) Try the latest brunch-unstable, I changed something which might have an impact on your device.
> 2) Reset chrome sync: Login to your Google dashboard account -> Go to the "Chrome" [...]

chronos@localhost / $ cat /var/log/messages |grep Android

2021-01-04T10:27:41.909621+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2729

2021-01-04T10:55:29.469582+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2729 stopped

2021-01-04T11:10:54.241831+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 7145

2021-01-04T11:13:39.733132+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 7145

2021-01-04T11:18:47.944227+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 7145 stopped

2021-01-04T11:19:41.503833+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 9535

2021-01-04T11:19:41.563832+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 9535

2021-01-04T11:20:14.386741+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 9535 stopped

2021-01-04T19:18:59.596225-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 1918

2021-01-04T19:26:52.041486-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 1918

2021-01-04T19:46:40.591138-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1872)] Android container with PID 1918 stopped

2021-01-04T19:47:17.081066-05:00 ERR session_manager[1176]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255

2021-01-04T19:47:17.731944-05:00 ERR session_manager[1176]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.

2021-01-04T19:47:17.732173-05:00 ERR session_manager[1176]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.

2021-01-04T20:30:03.232158-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 10903

2021-01-04T20:31:22.902470-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1872)] Android container with PID 10903 stopped

2021-01-04T20:32:38.571166-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 11838

2021-01-04T20:32:41.429059-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 11838

2021-01-04T20:48:07.822716-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1872)] Android container with PID 11838 stopped

2021-01-04T20:48:23.431907-05:00 ERR session_manager[10294]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255

2021-01-04T20:48:23.521759-05:00 ERR session_manager[10294]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.

2021-01-04T20:48:23.521945-05:00 ERR session_manager[10294]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.