sebanc/brunch #775: Self Repairing Chrome OS

yegender124001, 2021-01-04:
Hey, @sebanc today I start chrome os in my pc after a long time. Now I got a problem that all worked fine but after launching of brunch chrome os open. Then I got a loading and it says self-Repairing Chrome os. What should I do to recover It. I have installed Google hatch image in my pc

Thanks

sebanc, 2021-01-04:
Try to turn off fast startup in windows:
https://support.lenovo.com/gt/en/solutions/ht501793

yegender124001, 2021-01-04:
OK I will try it again. But the fast startup is turned off in my PC already

yegender124001, 2021-01-04:
Sebanc No success. I got this again.
[image]

Please help me

sebanc, 2021-01-04:
Boot to windows and press SHIFT while shutting down the computer. Not sure why this happens if fast startup is disabled though

yegender124001, 2021-01-04:
It leaves again there. No success again

sebanc, 2021-01-04:
Weird, try to run a checkdisk in Windows on the partition which contains the chromeos is image

yegender124001, 2021-01-04:
I have installed on my windows partition.  I am checking my windows partition by using chkdsk c:

yegender124001, 2021-01-04:
It stucks. 
[image]

ameyjoshi3068, 2021-01-04:
I would suggest you create a separate partition and move .img to that partition.
If you can create ext4 from Linux then go for it.
Don't forget to change your grub config according to the new partition.

yegender124001, 2021-01-04:
> I think this issue is most likely due to NTFS partition error(corruption).
> I would suggest you create a separate partition and move .img to that partition.
> If you can create ext4 from Linux then go for it.
> Don't forget to change your grub config according to the new partition.

It didn't stucked. I have connected PC to a remote session. So, It is a performance reason. I have do it again and It run without sticking.So, my Windows partition is not corrupted

ameyjoshi3068, 2021-01-05:
@yegender124001 Its good to know your partition was not the issue.
If I was in the place of you, I would have replaced all files from recovery image and brunch to .img.
So that any issue left behind get solved instead of waiting here for solution.

yegender124001, 2021-01-05:
OK I will try to copy it. If the issue remains then I will write here

closed 2021-01-09 by yegender124001