sebanc/brunch #778: Asus flip c302 audio issue

AminM77, 2021-01-05:
I recently installed windows10 on my chromebook asus c302. touchscreen and audio completely don't determine by windows and installing any driver related to skylake.
I installed chrome OS as second OS but still have audio issue, no speakers, no mic.
I tried brunch v85, 86, 87 with same rammus and latest cave version related to my chromebook.
cave_recovery doesnt open camera.
I also tried "options=asus_c302,internal_mic_fix,..." nothing happened.

sebanc, 2021-01-09:
Normally, "options=asus_c302" should fix audio but I forgot to test it on 5.4 kernel. Could you try to replace "/kernel" by "/kernel-4.19" on the kernel command line ?

AminM77, 2021-01-09:
How should I do that? I wrote "options=..." on crosh> shell. How could use kernel command line?

sebanc, 2021-01-10:
Refer to #603

AminM77, 2021-01-10:
It's also okay with kernel 5.4. kernel-4.19 not found. And I don't know which annoying clicks are you talking about.
Works fine by brunch_r86_k5.4_stable_20201201 and rammus v87.
But "crosh> battery_firmware info", nothing happen on any versions.

It would be great if audio works on linux and windows(+touchscreen). I'm still testing related drivers.
Thanks a lot.

AminM77, 2021-01-18:
One another problem...
due to less main eMMC storage (about 64G) that I installed Windows 10, I'm used to add a 64GB SD XC U3 and installed Chrome OS on it (SD card read & write speed is as fast enough as eMMC storage), but when I successfully login to my account, after a moments it got a black screen. sometimes I touch and move chrome browser window to prevent getting black. Is there any command for keep wake the screen up?
Notice that on first login(after first brunch builds, or after grub changes which brunch builds again) no black screen appears.