sebanc/brunch #779: Audio Not working with Eve/Pixelbook

ericksontx, 2021-01-05:
Moving this info here where it belongs from the Crunch issue board.

Brunch on Eve has two major issues, backlight control and audio. Backlight has been somewhat addressed (though screen is dim compared to normal max bright) but audio remains a problem - no audio whatsoever.

From what Sebanc said in a different thread, this is likely due to the requisite files being missing. See post pasted below. How can I pull these and build an updated/patched kernel that will have working audio? Thanks

"Ok for all on this thread. At least as far as sound related files go, I searched within https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/factory-eve-9667.B-chromeos-4.4 and the overall 9667 eve repository, downloaded the Kernel, Coreboot (factory), and Coreboot (firmware) tarballs and did a windows file search.

The relevant sound chipsets are Realtek RT5663, RT5514, and Maxim MAX98927, the three are used together as a combo config with the Intel Kaby Lake / Skylake architecture.

Searching for each of those chip model numbers individually yields the following file locations of files with the string in the filename/path, or mentioned within the text of the file itself. (long list below). Ignore the C:/eve/ as that is just where I unpacked the three tarballs to my hard drive.

Sebanc - how should I pull the correct files for copying into firmware and building a patched kernel? Mount the Eve 85 recovery image and copy them out of there?

(or, anyone else want to volunteer?)

C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\max98927.c
C:\eve\corebootfactory\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfactory\src\mainboard\google\eve\mainboard.c
C:\eve\corebootfactory\src\drivers\i2c\max98927\chip.h
C:\eve\corebootfactory\src\drivers\i2c\max98927\Kconfig
C:\eve\corebootfactory\src\drivers\i2c\max98927\Makefile.inc
C:\eve\corebootfactory\src\drivers\i2c\max98927\max98927.c
C:\eve\kernel\sound\soc\intel\boards\kbl_rt5663_max98927.c
C:\eve\kernel\sound\soc\intel\boards\kbl_rt5663_rt5514_max98927.c
C:\eve\kernel\sound\soc\codecs\max98927.c
C:\eve\kernel\sound\soc\codecs\max98927.h
C:\eve\kernel\Documentation\devicetree\bindings\sound\max9892x.txt
C:\eve\corebootfactory\src\drivers\i2c\max98927
C:\eve\corebootfirmware\src\drivers\i2c\max98927
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\Makefile.inc
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\max98927.c
C:\eve\corebootfirmware\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfirmware\src\mainboard\google\eve\mainboard.c
C:\eve\corebootfirmware\src\drivers\i2c\max98927\chip.h
C:\eve\corebootfirmware\src\drivers\i2c\max98927\Kconfig
C:\eve\corebootfirmware\src\drivers\i2c\max98927\Makefile.inc
C:\eve\corebootfirmware\src\drivers\i2c\max98927\max98927.c
C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\Makefile.inc
 
C:\eve\kernel\sound\soc\intel\boards\kbl_rt5663_max98927.c
C:\eve\kernel\sound\soc\intel\boards\kbl_rt5663_rt5514_max98927.c
C:\eve\kernel\sound\soc\codecs\rt5663.c
C:\eve\kernel\sound\soc\codecs\rt5663.h
C:\eve\kernel\sound\soc\intel\skylake\skl.c
C:\eve\kernel\Documentation\devicetree\bindings\sound\rt5663.txt
C:\eve\corebootfirmware\src\drivers\i2c\rt5663
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\Makefile.inc
C:\eve\corebootfirmware\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\rt5663.c
C:\eve\corebootfirmware\src\mainboard\google\eve\mainboard.c
C:\eve\corebootfirmware\src\drivers\i2c\rt5663\chip.h
C:\eve\corebootfirmware\src\drivers\i2c\rt5663\Kconfig
C:\eve\corebootfirmware\src\drivers\i2c\rt5663\Makefile.inc
C:\eve\corebootfirmware\src\drivers\i2c\rt5663\rt5663.c
C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\Makefile.inc
C:\eve\corebootfactory\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\rt5663.c
C:\eve\corebootfactory\src\mainboard\google\eve\mainboard.c
 
C:\eve\kernel\sound\soc\codecs\rt5514-spi.h
C:\eve\kernel\include\sound\rt5514.h
C:\eve\kernel\Documentation\devicetree\bindings\sound\rt5514.txt
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\Makefile.inc
C:\eve\corebootfirmware\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfirmware\src\soc\intel\skylake\nhlt\rt5514.c
C:\eve\corebootfirmware\src\mainboard\google\eve\mainboard.c
C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\Makefile.inc
C:\eve\corebootfactory\src\soc\intel\skylake\include\soc\nhlt.h
C:\eve\corebootfactory\src\soc\intel\skylake\nhlt\rt5514.c
C:\eve\corebootfactory\src\mainboard\google\eve\mainboard.c
C:\eve\kernel\sound\soc\intel\boards\kbl_rt5663_rt5514_max98927.c
C:\eve\kernel\sound\soc\codecs\rt5514.c
C:\eve\kernel\sound\soc\codecs\rt5514.h
C:\eve\kernel\sound\soc\codecs\rt5514-spi.c"

"Also, seems there is a mention in of this 3 chip combo in
corebootfactory\src\mainboard\google\eve\devicetree.cb as well

And probably the complete contents of this directory is a must-include:
Kernel - / sound / soc / intel / skylake"

sebanc, 2021-01-09:
Here you are looking at your bootloader which is not what I meant by firmware files.

Most of the firmware files are located in the recovery image partition 3 (which can only be mounted read only), usually at the following places:
- /lib/firmware
- /usr/share/alsa/ucm

So that I can add support for Pixelbook sound, you need to try identify the missing firmware files , copy them to the brunch rootfs and checking if sound start working after a reboot.

If you list the firmware files which are needed for Pixelbook, I can probably add support for it.

olm3ca, 2021-01-22:
Hi - I'm interested in helping to get this fixed. Currently running 86 on the Pixelbook via UEFI boot. 

In /usr/share/alsa/ucm I located the attached. I didn't find anything of interest in /lib/firmware, but I may be missing something. The full contents of that folder are included in firmware.txt also attached. 

As for next steps, I will try to add the firmware files I've found so far to brunch rootfs to see if that does anything. 

[firmware.txt](https://github.com/sebanc/brunch/files/5854151/firmware.txt)

[kbl_r5514_5663_max.zip](https://github.com/sebanc/brunch/files/5854135/kbl_r5514_5663_max.zip)

olm3ca, 2021-01-22:
Well, quick update - that didn't suffice. There must be something else missing. 

One thought: In Yusef Napora's guide to getting audio working on Ubuntu a few years ago, he developed ansible scripts that pull the drivers from the right locations. I'm not well-versed in how ansible works but will try reviewing them to see if there are any clues. Here are his notes, and within that repo, the audio scripts he uses: 

https://github.com/yusefnapora/pixelbook-linux/blob/master/implementation-details.md

sebanc, 2021-01-23:
Could you try the new release in brunch-unstable repository with the below additional steps:
- Replace "/kernel" in the grub configuration with "/kernel-test"
- add "options=native_chromebook_image" to the kernel command line

I would be interested to know if it helps with the sound issue even though this kernel used is 4.4 for now.

olm3ca, 2021-01-23:
@sebanc Audio works!! Wow - absolutely amazing, thank you. That did it... 

So I suppose this proves that the Pixelbook needs to remain on kernel 4.4 for audio to work, which makes sense honestly. Did the firmware files I provided help, or is this the only fix?

sebanc, 2021-01-23:
To be honest, I was already working on the same issue with someone in the discussions tab for pixelbook go:
https://github.com/sebanc/brunch/discussions/792
However, I would rather keep both discussions separate for now.

Globally, according to https://github.com/flantel/pixelbook-linux I knew that audio could work on 4.4 kernel, so I added it and created an option to keep the original firmwares.

This is just the first build that is working but I am still working on it and will try to see if I can somehow switch to a more recent kernel.

sebanc, 2021-01-23:
I had to change the brunch-unstable release, the steps with the new one are still the same but the name of the kernel is different:
- Replace "/kernel" in the grub configuration with "/kernel-chromebook"
- Add "options=native_chromebook_image" to the kernel command line

olm3ca, 2021-01-23:
Just in case this helps - I believe the kernel was updated from 4.4 to 5.4 in beta 87. So this is still very new, and the latest recovery is only 86... Would it make sense to wait until a new recovery drops, and then test? I'm quite content with 4.4, with full functionality.

olm3ca, 2021-01-24:
@sebanc it appears tablet mode isn't functioning. In tablet mode, on-screen keyboard doesn't pop up unless I enable on-screen keyboard manually, but then it's on all the time. Is this something you can fix in a subsequent release? Thanks!

sebanc, 2021-01-24:
Ah yes forgot about this one, it is an issue with MrChromebox firmware when used with kernel < 4.19... The only fix is to update the kernel to 4.19 or above which is actually in my plans but it might take time.

olm3ca, 2021-01-24:
ok, good to know. It can wait! If there's anything I can test, when you're ready, just let me know.

sebanc, 2021-01-24:
Sure, I will let you know when I attempt to switch to a more recent kernel

olm3ca, 2021-01-27:
@sebanc - looks like the latest unstable is booting with 5.4.88-brunch-sebanc-chromebook kernel (when using /kernel-chromebook in grub). So audio is not working again. Which version should we use on the Pixelbook?

Also, the latest recovery dropped, so we can use 87 now as long as that will work ok.

sebanc, 2021-01-28:
@olm3ca I tried to make audio work on 5.4 kernel in the last few brunch-unstable but I think I have identified the source of the sound problem on recent kernels and it is very difficult to fix so I reverted to 4.4 kernel in the latest brunch-unstable.

Nevertheless, the good news is that I think I found a fix for your tablet mode issue. Let me know if it works.

olm3ca, 2021-01-28:
@sebanc that's great, I'll test it and report back. Do I continue to use /kernel-test or /kernel-chromebook?

sebanc, 2021-01-28:
Same as previously:
- Replace "/kernel" in the grub configuration with "/kernel-chromebook"
- Add "options=native_chromebook_image" to the kernel command line

olm3ca, 2021-03-02:
@sebanc - not sure if you saw my question in discord, but just reporting here that if I add updates to grub by changing 
- options=native_chromebook_image to 
- options=enable_updates,native_chromebook_image

I can update Chrome OS, but it breaks audio. uname -a still reports 4.4.250-brunch-sebanc-chromebook though. If we could install updates without breaking audio that would be nice but I just wanted to see if it's something worth trying to figure out.

ericksontx, 2021-05-22:
> @olm3ca I tried to make audio work on 5.4 kernel in the last few brunch-unstable but I think I have identified the source of the sound problem on recent kernels and it is very difficult to fix so I [...]

@olm3ca - any further success with eve audio + brunch?

olm3ca, 2021-05-22:
@ericksontx just that I got it working and was quite happy with it... It was fully functional last I tried about a month or so ago. Details on the install are all on my repo. https://github.com/olm3ca/Pixelbook-Campfire

FWIW, I did hear recently that only the speakers were working and not headphones, but you should try it. I don't have a pixelbook anymore otherwise I'd test.

ericksontx, 2021-05-22:
> @ericksontx just that I got it working and was quite happy with it... It was fully functional last I tried about a month or so ago. Details on the install are all on my repo. [...]

This is terrific, great news and great work! You listed eve recovery v87, but is the operation similar to standard brunch in that brunch version and recovery versions known to be compatible, will work? (kernel 4.4 mod of course) So Brunch Unstable v90 with recovery 87/88/89/90? Thanks

olm3ca, 2021-05-23:
You have two options, you could either follow the directions for 87 and then update to 90, or try using brunch-unstable 90 with recovery 90. The way I know will work is the first way. Just make sure the grub entry has everything I used.

ericksontx, 2021-05-29:
> You have two options, you could either follow the directions for 87 and then update to 90, or try using brunch-unstable 90 with recovery 90. The way I know will work is the first way. Just make sure the grub entry has everything I used.

Ok great got it working with the Eve 87 recovery image.

Audio output works but a few anomalies:

Once or twice audio output went out and required reboot to get it back. Not too concerned with this one.

Microphone does not seem to be active at all, any chat app, Zoom, Teams, Skype, or browser based versions (Zoom,Gmeet,Teams) do not have a working mic even though the apps do have the mic selectable. Teams and Skype  both give the message "speaker off" even though you can hear the call audio.

Confirmed hardware is ok- running windows, all calling clients either native x64 versions or browser versions work 100% with speaker and mic having full functionality.

Reflashed back to stock with eve recovery 87, same thing all functions no problem.

So non functioning mic and the "speaker off" message are with Brunch only. It's as if it has some type of speakerphone mode detected and  turned off? Odd

Any ideas what might cause the mic to be turned off using brunch and the legacy kernel? Thanks

olm3ca, 2021-05-31:
Was meaning to respond earlier on this - I think Sebanc is the only one to know how to fix these audio issues. Unfortunately I don't have any way of helping either, as I don't have a Pixelbook anymore. You can find him quite regularly on the Brunch discord server, might help to ask him there.