sebanc/brunch #788: Chrome OS freezes once desktop is loaded can move mouse but nothing else.

zenn87, 2021-01-07:
Hello, new to brunch been tinkering with it the past few days as I'm desperate to ditch windows and love chrome OS. Would love to use my thinkpad instead of shelling out for a chromebook for school. 

I have managed to finish the brunch install, used the samus image to build it. I can get into the chrome os setup screens ( entering my email , connecting to wifi and enabling assitant.etc then it freezes. If i go browse as guest I get the chrome window the task bar and i can move my mouse but my clicks dont seem to register.

My system is a lenovo thinkpad t410

Intel Core i5-520M
8 Gbs Of Ram
240 GB SSD 
Intel QM57 chipset

Hoping someone can help me please really dont have the cash from a chrome book as a student and dont want to have to use fyde os.....
--

sebanc, 2021-01-09:
I have other reports of this on 1st gen intel, it seems something broke on chromeos r83 update (r81 should still work fine).

Otherwise, you can try kernel 4.19 by replacing "/kernel" in the grub configuration by "/kernel-4.19".

Also, you could try the release in "brunch-unstable" repo.

zenn87, 2021-01-09:
Hi sebanc thanks for the reply , I fixed it on my own by Us ng a older image .

However I can't seem to enable Linux o get a error everytime even had this when I booted dude is for testing. Even have virtualization enabled in mono settings any ideas ? Also I can't seem to change to canary or Dec channel?

closed 2021-01-09 by zenn87

OneManOneMan, 2022-09-16:
@zenn87 I am having the same problem. Could you tell me how you fixed it?