sebanc/brunch #789: asus X201e,CPU is celeron,google play still can't be used,please help me fix

oukaitou2020, 2021-01-07:
I read some issues,get some messages,somebody can help fix the problem,thanks
chronos@localhost / $ cat /var/log/messages |grep Android
2021-01-04T10:27:41.909621+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2729
2021-01-04T10:55:29.469582+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2729 stopped
2021-01-04T11:10:54.241831+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 7145
2021-01-04T11:13:39.733132+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 7145
2021-01-04T11:18:47.944227+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 7145 stopped
2021-01-04T11:19:41.503833+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 9535
2021-01-04T11:19:41.563832+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 9535
2021-01-04T11:20:14.386741+00:00 INFO session_manager[1407]: [INFO:session_manager_impl.cc(1872)] Android container with PID 9535 stopped
2021-01-04T19:18:59.596225-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 1918
2021-01-04T19:26:52.041486-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 1918
2021-01-04T19:46:40.591138-05:00 INFO session_manager[1176]: [INFO:session_manager_impl.cc(1872)] Android container with PID 1918 stopped
2021-01-04T19:47:17.081066-05:00 ERR session_manager[1176]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2021-01-04T19:47:17.731944-05:00 ERR session_manager[1176]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.
2021-01-04T19:47:17.732173-05:00 ERR session_manager[1176]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2021-01-04T20:30:03.232158-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 10903
2021-01-04T20:31:22.902470-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1872)] Android container with PID 10903 stopped
2021-01-04T20:32:38.571166-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 11838
2021-01-04T20:32:41.429059-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 11838
2021-01-04T20:48:07.822716-05:00 INFO session_manager[10294]: [INFO:session_manager_impl.cc(1872)] Android container with PID 11838 stopped
2021-01-04T20:48:23.431907-05:00 ERR session_manager[10294]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2021-01-04T20:48:23.521759-05:00 ERR session_manager[10294]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.
2021-01-04T20:48:23.521945-05:00 ERR session_manager[10294]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.

I see message=starting Android container failed

sebanc, 2021-01-09:
Have you tried the release in the "brunch-unstable" repo ?

Otherwise you could try to reset Chrome sync but I am not sure it will help in this case:
Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

oukaitou2020, 2021-01-09:
> Have you tried the release in the "brunch-unstable" repo ?
> 
> Otherwise you could try to reset Chrome sync but I am not sure it will help in this case:
> Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

Yes, I tried latest brunch-unstable ,but Google play still can't be used, I want to know if celeron 847 is so old,then can't use Google play?

Lucaacer, 2021-03-08:
This issue is very similar to this one https://github.com/sebanc/brunch/issues/842

I have also tried to create and run an android container from crosh with run_oci with mixed results.

oukaitou2020, 2021-03-10:
hi,friends
Thanks for your mail,I want to read this issue,if the problem fixed,I will send mail to you.