sebanc/brunch #790: Android apps not working

pixlofc, 2021-01-07:
When I first installed chromeOS it worked, but then i rebooted and they just dont launch, it just shows a spinner on the icon.

khanhtranngoccva, 2021-01-07:
Use options=android_init_fix, or if worse, wait for a minute on the login screen after boot. Or log out and try again. EDIT: bruh it still fails to work.

khanhtranngoccva, 2021-01-07:
I have noticed that the Android container will not boot at all if this dmesg line hasn't spawned by the time you press enter after keying the password on the login UI.
[  633.176570] capability: warning: `main' uses 32-bit capabilities (legacy support in use)
So the solution should be like this, I think I may not have explained it well enough.
1. Make an init script in /etc/init (or whatever) that checks for this EXACT line. (for example: dmesg | grep "32-bit capabilities" )
2. After that, only when the line is detected can the init script that starts the login UI execute.

pixlofc, 2021-01-07:
what

b4iterdev, 2021-02-10:
In short, it's because of your device, I doubt that you are booting from HDD or USB Drive, which mostly no chromebook uses.
So take a look at issue #813 , read my comment, or just grab an SSD so you won't get this error.
If your problem have solved, consider close this cases ( I see a lot of people including me are asking for this lol ), so close your case so dev can focus on another issue.

closed 2022-04-01 by pixlofc