sebanc/brunch #801: Lenovo T440s - 3 Finger Gestures

jaymanp, 2021-01-09:
Hello,

On ChromeOS 87 and only one issue so far... 3 finger gestures don't appear to be working. They are interpreted as two finger. I can scroll with either 2 or 3 finger. What's strange is at first boot post install, it worked. Since then, it has not. I've had similiar problems with Ubuntu using kernels 5.4 all the way to 5.10. Only way to fix it in Ubuntu was to reload the kernel module using "rmmod/modprobe psmouse"

Here is the udevadm output for the devices. I'm assuming there are two devices due to the trackpoint?
```
chronos@localhost / $ udevadm info /dev/input/mouse0
P: /devices/platform/i8042/serio1/input/input19/mouse0
N: input/mouse0
S: input/by-path/platform-i8042-serio-1-mouse
E: DEVLINKS=/dev/input/by-path/platform-i8042-serio-1-mouse
E: DEVNAME=/dev/input/mouse0
E: DEVPATH=/devices/platform/i8042/serio1/input/input19/mouse0
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_PATH=platform-i8042-serio-1
E: ID_PATH_TAG=platform-i8042-serio-1
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=32
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=17508423

chronos@localhost / $ udevadm info /dev/input/mouse1
P: /devices/platform/i8042/serio1/serio2/input/input20/mouse1
N: input/mouse1
S: input/by-path/platform-i8042-serio-2-mouse
E: DEVLINKS=/dev/input/by-path/platform-i8042-serio-2-mouse
E: DEVNAME=/dev/input/mouse1
E: DEVPATH=/devices/platform/i8042/serio1/serio2/input/input20/mouse1
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_INPUT_POINTINGSTICK=1
E: ID_PATH=platform-i8042-serio-2
E: ID_PATH_TAG=platform-i8042-serio-2
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=33
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=18319437
```

I'd appreciate any help on the matter! Thank you. This is a great product btw. I never imagined having ChromeOS on this device. Let me know if there is a way to contibute to your project (financially speaking).

jaymanp, 2021-01-09:
Sometimes on a fresh boot, three finger will work. Coming out of sleep, the trackpad will cease to function. Is there a way to be able to reload the touchpad driver?

jaymanp, 2021-01-09:
If I run these commands I can get the touchpad gestures working correctly (temporarily)

```
localhost /etc # rmmod psmouse
localhost /etc # modprobe psmouse
```

sebanc, 2021-01-09:
Could you try adding "psmouse.synaptics_intertouch=0" on the kernel command line ?

jaymanp, 2021-01-09:
> Could you try adding "psmouse.synaptics_intertouch=0" on the kernel command line ?

Hi Sebanc. That didn't work. I had that kernal option in there with =1 previously and that didn't work either. Once I manually reloaded the kernel module using rmmod and mobprobe after boot it works. This is the same behavior I had with 5.4, 5.8, 5.9, 5.10 kernels with Ubuntu.

jaymanp, 2021-01-09:
> Could you try adding "psmouse.synaptics_intertouch=0" on the kernel command line ?

I modified the kernel parameter accordingly (The right way this time) and still not working. 2 finger gestures are OK.... just doesn't recognize 3 finger. With the kernel parameter in place, manually reloading the module no longe restores the 3 finger gesture.

sebanc, 2021-01-09:
Could you try with psmouse.synaptics_intertouch=1 correctly set ?

jaymanp, 2021-01-09:
Sure....

It worked on the first boot. On reboot, the three finger gesture stopped working. This time reloading the psmouse module fixed it, but until it goes to sleep or is rebooted again.  

```
menuentry "ChromeOS" {
        regexp --set disk "(^.+)(,gpt)" $root
        linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug snd_hda_intel.model=headset-mic psmouse.synaptics_intertouch=1 \
                console= vt.global_cursor_default=0 brunch_bootsplash=default
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
```

sebanc, 2021-01-10:
Try to run the following commands in crosh shell:
```
sudo mkdir -p /var/brunch/bootscripts
echo "rmmod psmouse" | sudo tee /var/brunch/bootscripts/touchpad_fix.sh
echo "modprobe psmouse" | sudo tee -a /var/brunch/bootscripts/touchpad_fix.sh
sudo chmod 0755 /var/brunch/bootscripts/touchpad_fix.sh
```

Then reboot and see if it helps

jaymanp, 2021-01-10:
Do I need to add  #!/*bin*/*sh *to the script?

jaymanp, 2021-01-10:
> Try to run the following commands in crosh shell:
>
> ```
> sudo mkdir -p /var/brunch/bootscripts
> echo "rmmod psmouse" | sudo tee /var/brunch/bootscripts/touchpad_fix.sh
> echo "modprobe psmouse" | sudo [...]

No luck there. The problem is I need to do the rmmod and modprobe command multiple times to get it to work. Sometimes it's twice, sometimes 6 times for the gestures to fully restore. 

With Ubuntu, I needed to do the same.  

I do plan on replacing the touchpad in the coming days with one from a newer thinkpad. Perhaps it will land on a new driver and resolve this issue.

Thanks for your help. Let me know if you want me to try anything else.

jaymanp, 2021-01-11:
@sebanc 

I seem to have better luck by putting in a "sleep 5" between the rmmod and the modprobe. It still does not always work.

Is there a way to run a script coming out of sleep? I lose the touchpad entirely and running the reloading the module usually fixes it.

jaymanp, 2021-01-11:
I changed out the touchpad from the OEM 440s to the touchpad used by the 450/460 which is a three button. Gestures work as expected on boot and sleep. Thanks for your help @sebanc.

closed 2021-01-11 by jaymanp