sebanc/brunch #802: Can't get Linux VM to run

zenn87, 2021-01-09:
Hello I got chrome os to rub on my ThinkPad t410 using brunch and the r87 build. 

But for some reason I can't seem to run Linux apps. Every time I use the Linux beta setup tool of looks like it's going to install them it stops and says a error has occured while installing.

I read somewhere to try Google's name servers in DNS that didn't help , I also read to switch to Dev or canary. I tried that hit it says my account isn't the owner even though I've only ever signed in under my account.

I could use any guidance as linux is my last stop to being able to use this as my daily driver for university and work.

Thanks so much in advance

sebanc, 2021-01-09:
Is brunch installed on a usb flashdrive or SD card ? Usually this issue is due to low read/write speed.

If it's the case, I would suggest to try the dual boot method.

zenn87, 2021-01-09:
No I'm fully installed on a 1tb Samsung SSD

sebanc, 2021-01-09:
Your gpu being very old, you probably have to disable gpu acceleration in chrome://flags.

You could also try the latest brunch-unstable release as I tried to improve support for old gpus.

zenn87, 2021-01-09:
Ok thank you I'll try re install with brunch unstable. As I'm new to chrome os flags could you walk me through how-to do so quickly ? Again thank you

sebanc, 2021-01-09:
For flags, Open Chrome browser to the address "Chrome://flags" and disable the one called "Crostini GPU acceleration" or something similar.

"brunch-unstable" is a specific repo on my GitHub with test releases

zenn87, 2021-01-09:
Alright so I received installed with brunch unstable , however when I boot into to chrome and go into the flags nothing shows up it's all just blank.... Not a single flag shows ...

sebanc, 2021-01-09:
Ah yes forgot that, try to remove "virtio_gpu" from /etc/ui_use_flags.txt then reboot, it should do the same.

zenn87, 2021-01-09:
Sorry again new to all this how do I access that file from chrome os ?

sebanc, 2021-01-09:
In crosh shell (CTRL+ALT+T and type "shell" at the invite)
Then "sudo nano /etc/ui_use_flags.txt"
Remove that file and CTRL+X to save

zenn87, 2021-01-09:
That fixed it ! You sir are a legend , thank you !

mhbear14, 2021-01-20:
Close the issue if it's fixed @zenn87

closed 2021-01-20 by zenn87