sebanc/brunch #803: Issues with Headphones - Lenovo T440s

jaymanp, 2021-01-09:
@sebanc 

I'm having the same issue as [https://github.com/sebanc/brunch/issues/284](url). The headphones will not work unless you booth the machine with them connected.

I added this kernel option as suggested which fixed the problem for two others. It did not work for me.

`options=snd_hda_intel.model=headset-mic,psmouse.synaptics_intertouch=0`

Do you have any other suggestions?

sebanc, 2021-01-09:
Those 2 strings are not options but kernel parameters, add those 2 strings separated by a space before "options=..."

jaymanp, 2021-01-09:
How does this look? I took options out since I don't have any defined. I'll test it out once you confirm the kernel parameters are correct.

```
menuentry "ChromeOS" {
        regexp --set disk "(^.+)(,gpt)" $root
        linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug snd_hda_intel.model=headset-mic psmouse.synaptics_intertouch=0 \
                console= vt.global_cursor_default=0 brunch_bootsplash=default
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
```

sebanc, 2021-01-09:
It looks correct

jaymanp, 2021-01-09:
> It looks correct

Still not functional. Headphones register when connected, just no audio output.

jaymanp, 2021-01-10:
Using the USB port doesn't have this issue. I can buy an adapter for a few bucks so this is really a none issue now. Happy to debug further if you wish, otherwise, we can close it off.