sebanc/brunch #807: Sony SVP132A1CV, Brunch r87, usb-to-ethernet isn't recognize

Po-temkin, 2021-01-11:
Have some problems with usb3.0-to-ethernet adapter tp-link ue300.

daemongloom, 2021-01-11:
UE300 works fine for original Lenovo Chromebook Duet. Will test on HP Spectre X2  with brunch tomorrow.

Hardware info for ethernet adapter:
idVendor=2537, idProduct=0601, bcdDevice=30.0 
Driver r8152 loads "rtl8153a-3 v2 02/07/20" device.

Po-temkin, 2021-01-11:
Most interesting that the adapter's led blinking like it indicate a data transfer. But i can't found new link in setting or ipcofig.
If i turn off another link (wi-fi), the internet connection connection will be lost.
To "fix" this I reload the OS without adapter. It looks like that the karnel use wrong driver, but OS know the best way to work with.

sebanc, 2021-01-11:
Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and post dmesg.txt from your downloads folder)

daemongloom, 2021-01-12:
Works fine on HP Spectre X2. Chrome OS 87.0.4280.109, brunch r87 20201227.

Po-temkin, 2021-01-12:
[dmesg.txt](https://github.com/sebanc/brunch/files/5800923/dmesg.txt)

daemongloom, 2021-01-12:
What happens if you boot your laptop with any other usb device (flash disk, for example) in that usb port? Will it work after boot? Will usb ethernet adapter work if plugged to different usb port?
I can't see any software/driver issues in that log.

Po-temkin, 2021-01-12:
[dmesg1.txt](https://github.com/sebanc/brunch/files/5802883/dmesg1.txt)
I "paused" and booted with adapter... And it works. dmesg1.txt is log when all fine.