sebanc/brunch #808: tablet mode not triggered

Diliz, 2021-01-11:
Hello there,

First of all, thanks for the great work on brunch! That's really cool!

I actually have a teclast f5 convertible laptop, it works well with the options rtl8821ce and advanced_als, but it seems the tablet mode is not triggered when I rotate the lid as it should be, I tried the latest unstable brunch release as described [here](https://github.com/sebanc/brunch/issues/764#issuecomment-753292409) but it's not working :/

The chassis type is 31 and the [dmesg.txt](https://github.com/sebanc/brunch/files/5798038/dmesg.txt) as asked in the mentionned issue :)

Diliz, 2021-01-11:
Tried with tablet mode forced, auto rotate doesn't seem to be working as well :/

sebanc, 2021-01-13:
Sorry but regarding sensors your tablet does not use the common "intel sensor hub" which is the only driver that has been ported to work in chromeos (it uses "kxcjk_1013" module instead), someone with the same hardware would need to port this driver for it to work in chromeos.

Regarding tablet mode, teclast most likely use a custom method not supported in the linux kernel, you can use CTRL+ALT+SHIFT+T to switch to tablet mode but it is probably not necessary anyway if sensors are not supported.

Diliz, 2021-01-14:
Ah damn, thanks for replying, i'll check on another linux distribution or if someone found a driver/workaround for this one :)